lorol/LITTLEFS #1: When the file write data is full, the data is lost

LusterWong, 2020-08-14:
Thank you so much for porting Littlefs to Arduino. SPIFFS don't work very well for me, and as the free space gets smaller, the writing gets slower and slower.

Space allocation was 1.9MB APP 190KB SPIFFS.
My platform is ESP32, and I'm testing Littlefs in a very simple way, writing the file until it's full and then reading it out.

```
void appendFile(fs::FS &fs)
{
	uint32_t i = 0;
	uint32_t size;

	File file = fs.open(path, FILE_APPEND);
	if (!file)
	{
		Serial.println("- failed to open file for appending");
		return;
	}

	i = 0;
	while(1)
	{
		size = file.print("0123456789");
		if(size > 0)
		{
			i += 10;
			printf("used size = %d\n", LITTLEFS.usedBytes());
		}
		else
		{
			break;
		}
	}
	Serial.printf("i = %d\n", i);
	Serial.printf("\nfile size = %d\n", file.size());

	file.close();
}
```

This is a serial output:
```
............
............
used size = 376832
used size = 376832
used size = 376832
used size = 376832
D:\git\LITTLEFS\src\lfs.c:495:error: No more free space 53

i= 184060

file size = 0
```
I expect the variable "size" return less than 1 when the file is full and the data is written to the file system.
But the fact is not written, and file.size() returns 0. Because my application needs to keep writing data, it cannot write again when the file is full. Don't know how to solve this situation?

In addition, SPIFFS also appear in a similar situation, but worse is the endless cycle of writing without any return.I've also posted references here[https://github.com/espressif/arduino-esp32/issues/4256](url).

lorol, 2020-08-14:
To be written, the file needs to be closed before the system gets error. 
So depends of what you exactly target, you should take care to stop appending and close.

LusterWong, 2020-08-15:
> To be written, the file needs to be closed before the system gets error.
> So depends of what you exactly target, you should take care to stop appending and close.
> You are trying first to write, then to see what will happen. File system also engages sectors of the available space.

I changed the code to close the file and reopen it after each write.
```
void appendFile(fs::FS &fs)
{
        uint32_t i = 0;
	uint32_t size;

	File file;

	i = 0;
	while(1)
	{
		file = fs.open(path, FILE_APPEND);
		if (!file)
		{
			Serial.println("- failed to open file for appending");
			return;
		}

		size = file.print("0123456789");
		file.close();

		if(size > 0)
		{
			i += 10;
			printf("used size = %d\n", LITTLEFS.usedBytes());
		}
		else
		{
			break;
		}
	}

	Serial.printf("i = %d\n", i);
	Serial.printf("\nfile size = %d\n", file.size());

	file.close();
}
```
File writes become slow. When the file is full, the serial output "Error: No more free space 53", but the variable "size" is still > 0, can't get out of "break". An infinite loop and serial output.

I want to maximize the use of file space, and it can stop writing files when they are full. What should I do? 

Thank you.

lorol, 2020-08-15:
@LusterWong  Review first what you exactly need. On esp32 you may get it w/o a filesystem or even by using RAM only.
Just a simple buffering etc. Using files, involves caching and reserving larger than few bytes flash and ram memory chunks. 
You can try by file.flush() in the loop and do not not rely of file.print() to tell you if  bytes were not printed.
It doesn't seem to be an issue of LITTLEFS.

LusterWong, 2020-08-15:
> @LusterWong Review first what you exactly need. On esp32 you may get it w/o a filesystem or even by using RAM only.
> Just a simple buffering etc. Using files, involves caching and reserving larger [...]

I need to call.write() or.print() so that the data can be written to littlefs correctly and that I know when the space full.
Now calling.write() or.print(), the space will be full and the serial port will say "No more free space 53" but the return value will still be >0, so the write operation cannot be interrupted.

lorol, 2020-08-15:
Because write() or print() are not committing directly the changes to flash, they are "buffered".
Unfortunately, you cannot see available space while keep adding this way. System cannot tell you when to stop. 
Sorry, cannot help you.

LusterWong, 2020-08-15:
> Because write() or print() are not committing directly the changes to flash, they are "buffered".
> Unfortunately, you cannot see available space while keep adding this way. System cannot tell you when to stop.
> Sorry, cannot help you.

Should this be resolved? 

I am aware of the buffering problem. I have used littlefs in other MCUs and call close or flush every time data is written. If the space is full, the write function will return -28. 

But in arduino, the return value is always equal to the number of bytes written. Since the serial port prompts an error, why can't I get the return value of -28?

lorol, 2020-08-16:
Hi @LusterWong,
You can try this to see roughly what is happening behind the scenes and eventually get a workaround: 

```
void append2File(fs::FS &fs, const char * path){
    uint32_t i = 0;
    uint32_t y = 0;
    uint32_t z = 0;
    
    Serial.printf("Appending to file: %s\r\n", path);

    File file = fs.open(path, FILE_APPEND);
    if(!file){
        Serial.println("- failed to open file for appending");
        return;
    }

    y = (LITTLEFS.totalBytes() - LITTLEFS.usedBytes())/4096;
    
    printf("sectors available at start= %d\n", y );
    while(1){
        
        if (Serial.available() > 0) {
          break; // hit a key and enter to stop
           Serial.println("stopped by user");
        }

        z = y - i/4096;
        if (z>1){
          if(file.print("0123456789")){
              i += 10;
              printf("Added: %d bytes, sectors left:%d\n", i, z);
          } else {
              Serial.println(" - file print returned zero");
          }
        } else {
          Serial.println("- no enough sectors available");
        }
    }
    
    file.close();
    y = (LITTLEFS.totalBytes() - LITTLEFS.usedBytes())/4096;
    printf("sectors available at end= %d\n", y );
}

```

To confirm if this is specially an Arduino issue, you can try to build your esp32 app directly on ESP-IDF platform (not Arduino). Here is the SPIFFS example:
https://github.com/espressif/esp-idf/tree/master/examples/storage/spiffs
Then, you can use  https://github.com/joltwallet/esp_littlefs to adapt your app for LittleFS.

Back to Arduino world, there are thousands of (copied each-other) examples that actually use print() return value, like I did: 
https://github.com/lorol/LITTLEFS/blob/master/examples/LittleFS_test/LittleFS_test.ino#L100

Of course, these examples do not loop a cycle to keep appending like you wish. 

Keep in mind that LittleFS uses caching and sectors. So if you want to append even one byte beyond current sector, system will engage another sector, before yet knowing if it is physically available. The failure you see is at committing flush or close, means "too late" to go back and keep before that happened.
I cannot judge if this is wrong or right, if it is fixable or not. It requires much lower level that the wrappers I improvised on.

Hope this helps.

LusterWong, 2020-08-17:
> Hi @LusterWong,
> You can try this to see roughly what is happening behind the scenes and eventually get a workaround:
> (code below edited!)
> 
> ```
> void append2File(fs::FS &fs, const char * path){
>     uint32_t i = 0;
>     uint32_t y = 0;
>     uint32_t z = 0;
>     uint32_t v = 0;
>     
>     Serial.printf("Appending to file: %s\r\n", path);
> 
>     File file = fs.open(path, FILE_APPEND);
>     if(!file){
>         Serial.println("- failed to open file for appending");
>         return;
>     }
> 
>     y = (LITTLEFS.totalBytes() - LITTLEFS.usedBytes())/4096;
>     
>     printf("sectors available at start= %d\n", y );
>     v = y;
>     
>     while(1){
>         
>         if (Serial.available() > 0) {
>           break; // hit a key and enter to stop
>            Serial.println("stopped by user");
>         }
> 
>         z = y - i/4096;
>         if (z>0){
>           if(file.print("0123456789")){
>               i += 10;
>               printf("Added: %d bytes, sectors left:%d\n", i, z);
>               if (v > z) {
>                 file.flush();
>                 Serial.println("file flush");
>                 v = z;
>               }
>           } else {
>               Serial.println("file print returned zero");
>               break;
>           }
>         } else {
>           Serial.println("sectors are low");
>           break;
>         }
>     }
>     
>     file.close();
>     y = (LITTLEFS.totalBytes() - LITTLEFS.usedBytes())/4096;
>     printf("sectors available at end= %d\n", y );
> }
> ```
> 
> To confirm if this is specially an Arduino issue, you can try to build your esp32 app directly on ESP-IDF platform (not Arduino). Here is the SPIFFS example:
> https://github.com/espressif/esp-idf/tree/master/examples/storage/spiffs
> Then, you can use https://github.com/joltwallet/esp_littlefs to adapt your app for LittleFS.
> 
> Back to Arduino world, there are thousands of (copied each-other) examples that actually use print() return value, like I did:
> https://github.com/lorol/LITTLEFS/blob/master/examples/LittleFS_test/LittleFS_test.ino#L100
> 
> Of course, these examples do not loop a cycle to keep appending like you wish.
> 
> Keep in mind that LittleFS uses caching and sectors. So if you want to append even one byte beyond current sector, system will engage another sector, before yet knowing if it is physically available. The failure you see is at committing flush or close, means "too late" to go back and keep before that happened.
> I cannot judge if this is wrong or right, if it is fixable or not. It requires much lower level that the wrappers I improvised on.
> 
> Hope this helps.

Thank you for answering the question in such detail.

With respect to the return value of.print(), I replaced several board tests and found that the return value was equal to the number of bytes written, regardless of whether the space was full.

I understand the caching mechanism, which now calls.flush() after every write.Instead of using the return value of.print() to determine if the space is full, just check to see if the return value of file.size() has increased after each write.I'm not sure if that's right. I'm testing it right now.

lorol, 2020-08-17:
If you have only one (this) file open at same time by your application, I recommend while this file is "open",  you just estimate the space by numbers at opening and prevent to overfill by leaving enough bytes (say 2 x 4096) before the end instead of checking file.size(), print(), usedBytes() etc. while keep feeding data.  Filesystems do many low-level things, so the physical writing to the flash is not necessarily happening at given line of your code :) Imagine you have 5 files open at same time - it is not obvious how much space is available on every loop of yours.

LusterWong, 2020-08-18:
> If you have only one (this) file open at same time by your application, I recommend while this file is "open", you just estimate the space by numbers at opening and prevent to overfill by leaving [...]

The application samples 20 bytes of data every once in a while and saves it to Flash. When the space is full, it will no longer be written.Make sure that the data being written is actually written to Flash and not just a buffer.So I need to pay attention to the file system is full, stop writing in time.

"Open" all the time may not work, because the device may shut down at any time (after writing data), when no more than two files are open at the same time. :+1: 

Thank you for your detailed advice. I will continue the test.

lorol, 2020-08-19:
@LusterWong  if you open/close the file every time you need to append, by checking and leaving enough free space - it should work.
I am closing the case as nothing I can modify to accommodate your request and with the other FS (SPIFFS) results with your code are identical.

closed 2020-08-19 by lorol