lorol/LITTLEFS #2: Errors of open files

DmitryBorisenko33, 2020-08-17:
I have esp32 board. I using ESPAsyncWebServer and your little fs library. When i refresh web page ill get opening files errors. But page opening fully and working normally. Only errors in serial haunt me... 

E (18932) esp_littlefs: Failed to opendir "**/index.htm**". Error  (-2)
[E][vfs_api.cpp:64] open(): /littlefs/index.htm does not exist

E (20877) esp_littlefs: Failed to opendir "**/js/build.chart.js**". Error  (-2)
[E][vfs_api.cpp:64] open(): /littlefs/js/build.chart.js does not exist

E (20925) esp_littlefs: Failed to opendir "**/css/build.css**". Error  (-2)
[E][vfs_api.cpp:64] open(): /littlefs/css/build.css does not exist

E (21264) esp_littlefs: Failed to opendir "**/index.json.gz**". Error  (-2)
[E][vfs_api.cpp:64] open(): /littlefs/index.json.gz does not exist

E (21302) esp_littlefs: Failed to opendir "**/favicon.ico**". Error  (-2)
[E][vfs_api.cpp:64] open(): /littlefs/favicon.ico does not exist

E (21308) esp_littlefs: Failed to opendir "**/favicon.ico.gz**". Error  (-2)
[E][vfs_api.cpp:64] open(): /littlefs/favicon.ico.gz does not exist

E (21322) esp_littlefs: Failed to opendir "**/favicon.ico/index.htm**". Error  (-2)
[E][vfs_api.cpp:64] open(): /littlefs/favicon.ico/index.htm does not exist

E (21333) esp_littlefs: Failed to opendir "**/favicon.ico/index.htm.gz**". Error  (-2)
[E][vfs_api.cpp:64] open(): /littlefs/favicon.ico/index.htm.gz does not exist

E (21347) esp_littlefs: Failed to opendir "**/favicon.ico**". Error  (-2)
[E][vfs_api.cpp:64] open(): /littlefs/favicon.ico does not exist

E (21359) esp_littlefs: Failed to opendir "**/favicon.ico.gz**". Error  (-2)
[E][vfs_api.cpp:64] open(): /littlefs/favicon.ico.gz does not exist

E (21371) esp_littlefs: Failed to opendir "**/favicon.ico/index.htm**". Error  (-2)
[E][vfs_api.cpp:64] open(): /littlefs/favicon.ico/index.htm does not exist

E (21385) esp_littlefs: Failed to opendir "**/favicon.ico/index.htm.gz**". Error  (-2)
[E][vfs_api.cpp:64] open(): /littlefs/favicon.ico/index.htm.gz does not exist

NOT_FOUND: GET http://192.168.1.116/favicon.ico
_HEADER[Host]: 192.168.1.116
_HEADER[Connection]: keep-alive
_HEADER[Pragma]: no-cache
_HEADER[Cache-Control]: no-cache
_HEADER[Authorization]: Digest username="admin", realm="asyncesp", nonce="795d0e2e77f0bd1ac56d88a223e30cc8", uri="/favicon.ico", response="38eb1459f010bfbd37082bd9cf4a697d", opaque="0bc073c1bf61fa0ae678fa5892cfd2a6", qop=auth, nc=00000007, cnonce="4cc4260a509fc0f7"
_HEADER[User-Agent]: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/84.0.4147.125 Safari/537.36
_HEADER[Accept]: image/webp,image/apng,image/*,*/*;q=0.8
_HEADER[Referer]: http://192.168.1.116/
_HEADER[Accept-Encoding]: gzip, deflate
_HEADER[Accept-Language]: en-US,en;q=0.9,ru;q=0.8,mt;q=0.7
E (21476) esp_littlefs: Failed to opendir "**/config.live.json**". Error  (-2)
[E][vfs_api.cpp:64] open(): /littlefs/config.live.json does not exist

E (21483) esp_littlefs: Failed to opendir "**/config.live.json.gz**". Error  (-2)
[E][vfs_api.cpp:64] open(): /littlefs/config.live.json.gz does not exist

E (21497) esp_littlefs: Failed to opendir "**/config.live.json/index.htm**". Error  (-2)
[E][vfs_api.cpp:64] open(): /littlefs/config.live.json/index.htm does not exist

E (21511) esp_littlefs: Failed to opendir "**/config.live.json/index.htm.gz**". Error  (-2)
[E][vfs_api.cpp:64] open(): /littlefs/config.live.json/index.htm.gz does not exist

This files exist in flash: 

[image]

I not aware what reason of this problem. 

For declaration i using two lines:

```
#include "LITTLEFS.h"
#define LittleFS LITTLEFS
```

For initialization i using this line:    

`LittleFS.begin(false,"/littlefs", 10);`

In asinc web server i using object of little fs library to give acsess to web page files:

```
server.serveStatic("/css/", LittleFS, "/css/").setCacheControl("max-age=31536000");
server.serveStatic("/js/", LittleFS, "/js/").setCacheControl("max-age=31536000");
server.serveStatic("/favicon.ico", LittleFS, "/favicon.ico").setCacheControl("max-age=31536000");
server.serveStatic("/icon.jpeg", LittleFS, "/icon.jpeg").setCacheControl("max-age=31536000");

    server.serveStatic("/", LittleFS, "/")
        .setDefaultFile("index.htm")
        .setAuthentication(jsonReadStr(configSetupJson, "WebUser").c_str(), jsonReadStr(configSetupJson, "WebPass").c_str());
```

Please help me with this problem.

lorol, 2020-08-17:
Hi
You don't need to serveStatic again folders and files once you have it from "/" 
Remove leading slashes - LittleFS doesnt like them, try this: ```server.serveStatic("icon.jpeg", LittleFS, "/icon.jpeg").setCacheControl("max-age=31536000");```  Again, I don't see reason to serve again separately.

DmitryBorisenko33, 2020-08-17:
I'll check the version of your fork and then I'll post the results. I will also apply the above changes to the code.

DmitryBorisenko33, 2020-08-17:
I tried to do above changes and i tryed fork version. I have errors: 

[E][vfs_api.cpp:64] open(): /littlefs/index.htm does not exist
[E][vfs_api.cpp:64] open(): /littlefs/config.live.json does not exist
[E][vfs_api.cpp:64] open(): /littlefs/config.live.json.gz does not exist
[E][vfs_api.cpp:64] open(): /littlefs/config.live.json/index.htm does not exist
[E][vfs_api.cpp:64] open(): /littlefs/config.live.json/index.htm.gz does not exist

I tryed to use only one .serveStatic:

```
server.serveStatic("/", LittleFS, "/")
        .setDefaultFile("index.htm")
        .setAuthentication(jsonReadStr(configSetupJson, "WebUser").c_str(), jsonReadStr(configSetupJson, "WebPass").c_str());

```
But normally you need to use several **.serveStatic** to specify exactly which files to cache. Therefore, with one **.serveStatic** it will not be possible to configure caching correctly.

SPIFFS library with same code working without any errors.  Based on this, I think that the LittleFs library has differences from the SPIFFS library. I would like to ask you to do reverse support of this library and SPIFFS library. It is necessary to work with an **async web server** without errors.  I also wanted to thank you for your great job.

lorol, 2020-08-17:
Please provide the entire code and data folder and I can try to help you.  Of course, they are different filesystems and nothing to be reversed there.
If your code or part of it can run on esp8266 you can test it as well- there the LittleFS is part of the core support. 
I believe part of your issue is that first part of the path ** /littlefs/**config.live.json ...  doesn't match to where the files are.

DmitryBorisenko33, 2020-08-18:
[Link to all code with date folder](https://github.com/DmitryBorisenko33/Unikum) for platformio. 
Thank you very much for helping me.

lorol, 2020-08-19:
@DmitryBorisenko33 

**WebHandlers of ESPESPAsyncWebServer** check and take a normal or .gz file by results of calling **open()** from **FS vfs_api**
SPIFFS has no folders but returns true on opendir() - I believe this is done purposely to mitigate the SPIFFS limitations.
LittleFS has folders and returns false at this point. That's why you see log_e("%s does not exist", temp);
This is actually the purpose of the check - file does not exist, but on SPIFFS passes silently.
You can confirm if you add after line 59 of  **...libraries/FS/src/vfs_api.cpp:**
```
log_e("%s not a file, opens as dir:", temp);
```

On your project you can ignore these errors you see or make them disappeared by modifying  **...ESPAsyncWebServer/src/WebHandlers.cpp** this way:
```
bool AsyncStaticWebHandler::_fileExists(AsyncWebServerRequest *request, const String& path)
{
  bool fileFound = false;
  bool gzipFound = false;

  String gzip = path + ".gz";

  if (_gzipFirst) {
    if (_fs.exists(gzip)) { 
      request->_tempFile = _fs.open(gzip, "r");
      gzipFound = FILE_IS_REAL(request->_tempFile);
    }
    if (!gzipFound){
      if (_fs.exists(path)) { 
       request->_tempFile = _fs.open(path, "r");
       fileFound = FILE_IS_REAL(request->_tempFile);
      }
    }
  } else {
    if (_fs.exists(path)) {
     request->_tempFile = _fs.open(path, "r");
     fileFound = FILE_IS_REAL(request->_tempFile);
    }
    if (!fileFound){
     if (_fs.exists(gzip)) {
      request->_tempFile = _fs.open(gzip, "r");
      gzipFound = FILE_IS_REAL(request->_tempFile);
     }
    }
  }
```

lorol, 2020-08-19:
See also in the clone i did here https://github.com/lorol/Unikum for other suggestions.
An interesting web interface and code structure BTW. Thanks. I can use it as PIO tests base.

DmitryBorisenko33, 2020-08-27:
Thank you so match. I have already change **ESPAsyncWebServer/src/WebHandlers.cpp** and all working with out errors!