lorol/LITTLEFS #3: problem for first startup after flashing only firmware

ramidomalys, 2020-09-22:
hello and thank you so much for your rich contribution 
i am having a problem when i first flash a firmware ( not the partition image ) 
the thing is: each time i flash a new firmware to the flash,  i cant get the littlefs working in the first boot without a hardware reset , not for read/write/ nor format 
to produce the problem : 
- erase the flash
- flash firmware and partition table
- ( flash partition image or not : same result)
the only solution is to boot the firmware once: ( the little fs tries to format and fails), (tried to make software reboot via esp_restart and didnt work out) then do a hardware reset via the button . although i had something similar to this in SPIFFS (couldnt resolve it) before changing to LittleFS.
 here is the log where the firmware tries to format and access littlefs partition and force reboot when fails, until i do a hardware reset and then it works like a charm !  : 
any idea ? 

 rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7192
load:0x40078000,len:13100
load:0x40080400,len:3896
entry 0x40080688
␛[0;32mI (29) boot: ESP-IDF 3.40100.200827 2nd stage bootloader␛[0m
␛[0;32mI (29) boot: compile time 09:19:17␛[0m
␛[0;32mI (29) boot: chip revision: 3␛[0m
␛[0;32mI (33) boot_comm: chip revision: 3, min. bootloader chip revision: 0␛[0m
␛[0;32mI (40) boot.esp32: SPI Speed      : 40MHz␛[0m
␛[0;32mI (44) boot.esp32: SPI Mode       : DIO␛[0m
␛[0;32mI (49) boot.esp32: SPI Flash Size : 16MB␛[0m
␛[0;32mI (54) boot: Enabling RNG early entropy source...␛[0m
␛[0;32mI (59) boot: Partition Table:␛[0m
␛[0;32mI (63) boot: ## Label            Usage          Type ST Offset   Length␛[0m
␛[0;32mI (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000␛[0m
␛[0;32mI (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000␛[0m
␛[0;32mI (85) boot:  2 factory          factory app      00 00 00010000 00100000␛[0m
␛[0;32mI (92) boot:  3 ota_0            OTA app          00 10 00110000 00100000␛[0m
␛[0;32mI (100) boot:  4 ota_1            OTA app          00 11 00210000 00100000␛[0m
␛[0;32mI (107) boot:  5 wifi1            Unknown data     01 82 00310000 00100000␛[0m
␛[0;32mI (115) boot: End of partition table␛[0m
␛[0;32mI (119) boot_comm: chip revision: 3, min. application chip revision: 0␛[0m
␛[0;32mI (126) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x083b0 ( 33712) map␛[0m
␛[0;32mI (148) esp_image: segment 1: paddr=0x000183d8 vaddr=0x3ffb0000 size=0x021dc (  8668) load␛[0m
␛[0;32mI (152) esp_image: segment 2: paddr=0x0001a5bc vaddr=0x40080000 size=0x00404 (  1028) load␛[0m
␛[0;32mI (156) esp_image: segment 3: paddr=0x0001a9c8 vaddr=0x40080404 size=0x05650 ( 22096) load␛[0m
␛[0;32mI (174) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x22530 (140592) map␛[0m
␛[0;32mI (228) esp_image: segment 5: paddr=0x00042558 vaddr=0x40085a54 size=0x04790 ( 18320) load␛[0m
␛[0;32mI (242) boot: Loaded app from partition at offset 0x10000␛[0m
␛[0;32mI (242) boot: Disabling RNG early entropy source...␛[0m
␛[0;32mI (242) cpu_start: Pro cpu up.␛[0m
␛[0;32mI (246) cpu_start: Application information:␛[0m
␛[0;32mI (251) cpu_start: Project name:     draft␛[0m
␛[0;32mI (255) cpu_start: App version:      6d88366-dirty␛[0m
␛[0;32mI (261) cpu_start: Compile time:     Sep 22 2020 09:18:26␛[0m
␛[0;32mI (267) cpu_start: ELF file SHA256:  5a1f2dd8a49c75c6...␛[0m
␛[0;32mI (273) cpu_start: ESP-IDF:          3.40100.200827␛[0m
␛[0;32mI (278) cpu_start: Starting app cpu, entry point is 0x40081040␛[0m
␛[0;32mI (271) cpu_start: App cpu up.␛[0m
␛[0;32mI (289) heap_init: Initializing. RAM available for dynamic allocation:␛[0m
␛[0;32mI (296) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM␛[0m
␛[0;32mI (302) heap_init: At 3FFB2A70 len 0002D590 (181 KiB): DRAM␛[0m
␛[0;32mI (308) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM␛[0m
␛[0;32mI (315) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM␛[0m
␛[0;32mI (321) heap_init: At 4008A1E4 len 00015E1C (87 KiB): IRAM␛[0m
␛[0;32mI (327) cpu_start: Pro cpu start user code␛[0m
␛[0;32mI (345) spi_flash: detected chip: gd␛[0m
␛[0;32mI (346) spi_flash: flash io: dio␛[0m
␛[0;32mI (346) cpu_start: Starting scheduler on PRO CPU.␛[0m
␛[0;32mI (0) cpu_start: Starting scheduler on APP CPU.␛[0m
␛[0;32mI (6559) SPIFFS wifi1: mounting  SPIFFS partition wifi1....␛[0m
C:\Users\Rami_Domalys\.platformio\packages\framework-espidf\components\esp_littlefs\src\littlefs\lfs.c:998:error: Corrupted dir pair at {0x0, 0x1}
␛[0;33mW (6567) esp_littlefs: mount failed,  (-84). formatting...␛[0m
C:\Users\Rami_Domalys\.platformio\packages\framework-espidf\components\esp_littlefs\src\littlefs\lfs.c:1685:debug: Bad block at 0x0
[... 185 lines cut ...]
␛[0;32mI (308) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM␛[0m
␛[0;32mI (314) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM␛[0m
␛[0;32mI (321) heap_init: At 4008A1E4 len 00015E1C (87 KiB): IRAM␛[0m
␛[0;32mI (327) cpu_start: Pro cpu start user code␛[0m
␛[0;32mI (345) spi_flash: detected chip: gd␛[0m
␛[0;32mI (346) spi_flash: flash io: dio␛[0m
␛[0;32mI (346) cpu_start: Starting scheduler on PRO CPU.␛[0m
␛[0;32mI (0) cpu_start: Starting scheduler on APP CPU.␛[0m
␛[0;32mI (6559) SPIFFS wifi1: mounting  SPIFFS partition wifi1....␛[0m
C:\Users\Rami_Domalys\.platformio\packages\framework-espidf\components\esp_littlefs\src\littlefs\lfs.c:998:error: Corrupted dir pair at {0x0, 0x1}
␛[0;33mW (6567) esp_littlefs: mount failed,  (-84). formatting...␛[0m
␛[0;32mI (6917) SPIFFS wifi1: Partition size: total: 1024.0 KB, used: 8.0 KB , free: 1016.0 KB␛[0m
␛[0;32mI (6919) SPIFFS wifi1: Active partition is set to :(wifi1)␛[0m
␛[0;32mI (6922) SPIFFS wifi1: Mount OK␛[0m
␛[0;32mI (6926) SPIFFS wifi1: Opening file /w/config1.txt  for mode r ....␛[0m
␛[0;31mE (6934) SPIFFS wifi1: Failed to open file for r ␛[0m
␛[0;31mE (6938) WIFI Directory: No file found..␛[0m
␛[0;33mW (6943) WIFI Directory: Creating new file w/config1.txt ..  ␛[0m
␛[0;32mI (6949) SPIFFS wifi1: Opening file /w/config1.txt  for mode w+ ....␛[0m
␛[0;32mI (6963) SPIFFS wifi1: file /w/config1.txt is OPEN.␛[0m
␛[0;32mI (6963) WIFI Directory: file created succefully ...no entries over here, adding default entry ␛[0m
␛[0;32mI (6977) SPIFFS wifi1: File closed␛[0m
␛[0;32mI (6978) WIFI Directory:  Entry with ssid : DOM_18DD7F is added␛[0m

THANKS IN ADVANCE

lorol, 2020-09-22:
@ramidomalys 
1) Flash something else or “Erase Flash” (Arduino IDE doesn’t have direct way to erase the entire flash of ESP32,  AFAIK)
2) Create a /data subfolder in the folder where your .ino file is located and inside this /data folder, put some example text files for test.
3) Flash data partition automatically made from this folder by this tool: https://github.com/lorol/arduino-esp32fs-plugin
Select appropriate FS type
4) Upload the firmware

ramidomalys, 2020-09-23:
Hello @lorol  thank you so much for your kind reply, 
 the thing is , i am already using these steps, i am on platformio and using mklittlefs.exe and i can upload the data partition and the firmware with no problem, 
my problem is, in the very first boot, i can access the uploaded image for read with no problem, but the write operations always fails with some error in writing to flash/invalid flash write. OR formatting operation act the same ( in case i didn't upload an image as well) , i need to do a hardware reset once, then all things go to normal, i can format, read and write like a charm.
FYI : EXACTLY SAME attitude with SPIFFS. 

```

platfomio.ini : 
extra_scripts = LittleFSBuilder.py

```
```
(Import("env")
env.Replace (MKSPIFFSTOOL = env.get ("PROJECT_DIR") + '/mklittlefs.exe')
```

````
esp_vfs_littlefs_conf_t conf1;
    conf1.base_path = "/w";
    conf1.partition_label = "wifi1";
    conf1.format_if_mount_failed = true;
    //conf1.max_files=10;
    conf1.dont_mount = false;
    esp_err_t ret;
    ret = esp_vfs_littlefs_register(&conf1);
    if (ret != ESP_OK)
    {
        if (ret == ESP_FAIL)
        {
            ESP_LOGW("TAG", "Failed to mount ..formatting ");
            if (esp_littlefs_format(conf1.partition_label) == ESP_OK)
            {
                ESP_LOGI("TAG", "Format success ..trying to mount");
                ret = esp_vfs_littlefs_register(&conf1);
                if (ret != ESP_OK)
                {
                    ESP_LOGE(TAG, "format error ..");
                }
                ESP_LOGI(TAG, "mount success ..");
            }
            else
            {
                ESP_LOGE(TAG, "Mount and format Failed ,check partition table...rebooting : this might be the first firmware boot after erase...");
                //esp_restart();
            }
        }
        else
        {
            ESP_LOGE(TAG, "Failed to mount SPIFFS (%s)", esp_err_to_name(ret));
        }
    }
    else
    {
        bool isMounted = esp_littlefs_mounted(conf1.partition_label); // succefull mounting?
        if (!isMounted)
        {
            ESP_LOGE(TAG, "Mount or format fail after verification ...");
        }
        size_t total;
        size_t used;
        ret = esp_littlefs_info(conf1.partition_label, &total, &used);
        if (ret != ESP_OK)
        {
            //some kind of error occured (probably partiotion name
            ESP_LOGE(TAG, "Failed to get SPIFFS partition information (%s)", esp_err_to_name(ret));
        }
        ESP_LOGI(TAG, "Partition size: total: %.1f KB, used: %.1f KB , free: %.1f KB", ((double)total / 1024), ((double)used / 1024), (((double)total - (double)used) / 1024));
        ESP_LOGI(TAG, "Active partition is set to :(%s)", conf1.partition_label);
        ESP_LOGI(TAG, "Mount OK");
```
this sketch is used without uploading image.
(i try to format even two times here : automatic format if fails that is embedded with the register function, and also individually)

so after a hardware reset format success. 

another sketch with exact same code, but with uploading image, mount success , read files success, but write file, remove file, even format , fails in the same way, need one hardware reset .... 

THANK YOU SO MUCH FOR YOUR CONTRIBUTION

ramidomalys, 2020-09-23:
this is the log after i uploaded image : i can read file and get my wifi list no problem 

```

I (7549) SPIFFS wifi2: mounting  SPIFFS partition wifi2....
I (7556) SPIFFS wifi2: Partition size: total: 512.0 KB, used: 20.0 KB , free: 492.0 KB
I (7560) SPIFFS wifi2: Active partition is set to :(wifi2)
I (7566) SPIFFS wifi2: Mount OK
I (7568) SPIFFS wifi2: Opening file /w/config1.txt  for mode r ....
I (7575) SPIFFS wifi2: file /w/config1.txt is OPEN.
I (7588) SPIFFS wifi2: File closed
I (7588) WIFI Directory: directory is loaded and initialized with 19 entries 
 list size is : 19
entry: 0|  ssid : DOMACERTIF2 | Pass: DOMACERTIF2 | Priority : 250 | fail count : 0 | last connection status FAIL | permanent yes 
entry: 1|  ssid : DOMACERTIF2 | Pass: DOMACERTIF2 | Priority : 250 | fail count : 0 | last connection status FAIL | permanent yes
entry: 2|  ssid : DOMACERTIF2 | Pass: DOMACERTIF2 | Priority : 250 | fail count : 0 | last connection status FAIL | permanent yes 
entry: 3|  ssid : DOMACERTIF2 | Pass: DOMACERTIF2 | Priority : 250 | fail count : 0 | last connection status FAIL | permanent yes 
entry: 4|  ssid : DOMACERTIF2 | Pass: DOMACERTIF2 | Priority : 250 | fail count : 0 | last connection status FAIL | permanent yes
entry: 5|  ssid : DOMACERTIF2 | Pass: DOMACERTIF2 | Priority : 250 | fail count : 0 | last connection status FAIL | permanent yes 
entry: 6|  ssid : DOMACERTIF | Pass: DOMACERTIF | Priority : 50 | fail count : 0 | last connection status FAIL | permanent yes 
entry: 7|  ssid : DOMACERTIF1 | Pass: DOMACERTIF1 | Priority : 50 | fail count : 0 | last connection status FAIL | permanent yes
entry: 8|  ssid : DOMACERTIF | Pass: DOMACERTIF | Priority : 50 | fail count : 0 | last connection status FAIL | permanent yes
entry: 9|  ssid : DOMACERTIF1 | Pass: DOMACERTIF1 | Priority : 50 | fail count : 0 | last connection status FAIL | permanent yes 
entry: 10|  ssid : DOMACERTIF | Pass: DOMACERTIF | Priority : 50 | fail count : 0 | last connection status FAIL | permanent yes
entry: 11|  ssid : DOMACERTIF1 | Pass: DOMACERTIF1 | Priority : 50 | fail count : 0 | last connection status FAIL | permanent yes
entry: 12|  ssid : DOMACERTIF | Pass: DOMACERTIF | Priority : 50 | fail count : 0 | last connection status FAIL | permanent yes 
entry: 13|  ssid : DOMACERTIF1 | Pass: DOMACERTIF1 | Priority : 50 | fail count : 0 | last connection status FAIL | permanent yes
entry: 14|  ssid : DOMACERTIF | Pass: DOMACERTIF | Priority : 50 | fail count : 0 | last connection status FAIL | permanent yes 
entry: 15|  ssid : DOMACERTIF1 | Pass: DOMACERTIF1 | Priority : 50 | fail count : 0 | last connection status FAIL | permanent yes
entry: 16|  ssid : DOMACERTIF | Pass: DOMACERTIF | Priority : 50 | fail count : 0 | last connection status FAIL | permanent yes 
entry: 17|  ssid : DOMACERTIF1 | Pass: DOMACERTIF1 | Priority : 50 | fail count : 0 | last connection status FAIL | permanent yes 
entry: 18|  ssid : Admin-Domalys | Pass: D0m@lys240 | Priority : 1 | fail count : 1 | last connection status PASS | permanent yes
```

and this is the log when i try to modify the file just when f open is called  : 
```

I (12513) WIFI Directory: saving wifi list ...
C:\Users\Rami_Domalys\.platformio\packages\framework-espidf\components\esp_littlefs\src\littlefs\lfs.c:1685:debug: Bad block at 0x0
C:\Users\Rami_Domalys\.platformio\packages\framework-espidf\components\esp_littlefs\src\littlefs\lfs.c:1691:warn: Superblock 0x0 has become unwritable
I (12548) WIFI Directory: FIle Open for
C:\Users\Rami_Domalys\.platformio\packages\framework-espidf\components\esp_littlefs\src\littlefs\lfs.c:2635:debug: Bad block at 0xe
C:\Users\Rami_Domalys\.platformio\packages\framework-espidf\components\esp_littlefs\src\littlefs\lfs.c:2635:debug: Bad block at 0xf
C:\Users\Rami_Domalys\.platformio\packages\framework-espidf\components\esp_littlefs\src\littlefs\lfs.c:2635:debug: Bad block at 0x10
C:\Users\Rami_Domalys\.platformio\packages\framework-espidf\components\esp_littlefs\src\littlefs\lfs.c:2635:debug: Bad block at 0x11
C:\Users\Rami_Domalys\.platformio\packages\framework-espidf\components\esp_littlefs\src\littlefs\lfs.c:2635:debug: Bad block at 0x12
C:\Users\Rami_Domalys\.platformio\packages\framework-espidf\components\esp_littlefs\src\littlefs\lfs.c:2635:debug: Bad block at 0x13
C:\Users\Rami_Domalys\.platformio\packages\framework-espidf\components\esp_littlefs\src\littlefs\lfs.c:2635:debug: Bad block at 0x14
C:\Users\Rami_Domalys\.platformio\packages\framework-espidf\components\esp_littlefs\src\littlefs\lfs.c:2635:debug: Bad block at 0x15
C:\Users\Rami_Domalys\.platformio\packages\framework-espidf\components\esp_littlefs\src\littlefs\lfs.c:2635:debug: Bad block at 0x16
C:\Users\Rami_Domalys\.platformio\packages\framework-espidf\components\esp_littlefs\src\littlefs\lfs.c:2635:debug: Bad block at 0x17
C:\Users\Rami_Domalys\.platformio\packages\framework-espidf\components\esp_littlefs\src\littlefs\lfs.c:2635:debug: Bad block at 0x18
C:\Users\Rami_Domalys\.platformio\packages\framework-espidf\components\esp_littlefs\src\littlefs\lfs.c:2635:debug: Bad block at 0x19
C:\Users\Rami_Domalys\.platformio\packages\framework-espidf\components\esp_littlefs\src\littlefs\lfs.c:2635:debug: Bad block at 0x1a
C:\Users\Rami_Domalys\.platformio\packages\framework-espidf\components\esp_littlefs\src\littlefs\lfs.c:2635:debug: Bad block at 0x1b
[... cut ...]
I (21146) WIFI Directory: ssid DOMACERTIF1 is removed
I (21146) WIFI Directory: saving wifi list ...
I (21152) WIFI Directory: FIle Open for
I (21180) SPIFFS wifi2: File closed
I (36691) WS-Actions-WifiManager: Got command for get ssid
I (43658) WS-Actions-WifiManager: Got command for ssid=>DOMACERTIF
I (43659) WIFI Directory: ssid DOMACERTIF is removed
I (43659) WIFI Directory: saving wifi list ...
I (43669) WIFI Directory: FIle Open for
I (43674) SPIFFS wifi2: File closed
```
all seems good, opening file, modifying an closing it is all good ...

lorol, 2020-09-23:
If you have similar behaviour with SPIFFS and LITTLEFS both, then the problem is in your application.
Try to use the default partition table for test because on 2-step uploading of data and firmware, especially with custom partition scheme it may overwrite.

ramidomalys, 2020-09-23:
thanks for the reply mate, i tried another ESP32 and it worked like a charm .... i guess the one i had have a faulty flash even though its brand new .....

ramidomalys, 2020-09-23:
apparently, i did reproduced the problem on another esp32 ,  but i am on WROVER-16MB
i couldn't produce the problem at all on ESP32-WROOM 4MB ......
there is something wrong with the WROOVER or its 16MB FLASH ( tried on two modules already )....

lorol, 2020-09-24:
For 16MB Flash module, did you specify in platformIo.ini  appropriate partition csv file?
Example:
board_build.partitions = partitions_16custom.csv
...
And have the csv file in sane directory.

ramidomalys, 2020-09-24:
I am not sure what i am missing in the partition CSV File, of course i modified it and i have something like this 
board_build.partitions = partitions_custom.csv
present in the project directory . 

here is the partition table in he csv file 
 # Name,   Type, SubType, Offset,  Size, Flags
# Note: if you have increased the bootloader size, make sure to update the offsets to avoid overlap
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,    0xf000 ,   0x1000,
factory,  app,  factory, ,        1M,
ota_0,    app,  ota_0,   ,        1M,
ota_1,    app,  ota_1,   ,        1M,
wifi2,  data ,spiffs ,   , 512k,

lorol, 2020-09-25:
I uploaded PlatformIO example. Could you please test it with custom csv on 16M board and report?

ramidomalys, 2020-09-25:
i am working on esp-idf framework , but still , just tested you example, it gave same exact behavior , completely failed first boot , sometimes the bad block log just keep going for couple hundreds of lines ......
 `

```
PS C:\Users\Rami_Domalys\Downloads\LITTLEFS-master\LITTLEFS-master\examples\LITTLEFS_PlatformIO> & C:/Users/Rami_Domalys/.platformio/penv/Scripts/Activate.ps1
(penv) PS C:\Users\Rami_Domalys\Downloads\LITTLEFS-master\LITTLEFS-master\examples\LITTLEFS_PlatformIO> pio run -t uploadfs   
Processing esp32 (platform: https://github.com/platformio/platform-espressif32.git; board: esp32dev; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (2.0.0+sha.3688b7b) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10004.200129 (1.0.4)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <LittleFS_esp32> 1.0.0+sha.f6f365f
|   |-- <FS> 1.0
|-- <FS> 1.0
Building in release mode
Building SPIFFS image from 'data' directory to .pio\build\esp32\spiffs.bin
/file1.txt
/testfolder/test2.txt
Looking for upload port...
Auto-detected: COM6
Uploading .pio\build\esp32\spiffs.bin
esptool.py v2.6
Serial port COM6
Connecting......
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: b8:f0:09:93:63:7c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 16MB
Compressed 3604480 bytes to 3923...

Writing at 0x00c90000... (100 %)
Wrote 3604480 bytes (3923 compressed) at 0x00c90000 in 0.1 seconds (effective 299558.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00  
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
�Listing directory: /
  FILE: /file1.txt      SIZE: 3
  FILE: /test.txt       SIZE: 0
  DIR : /testfolder
Creating Dir: /mydir
.pio\libdeps\esp32\LittleFS_esp32\src\lfs.c:1685:debug: Bad block at 0xe9
.pio\libdeps\esp32\LittleFS_esp32\src\lfs.c:1685:debug: Bad block at 0xeb
[... 387 lines cut ...]
Writing file: /hello.txt
- file written
Appending to file: /hello.txt
- message appended
Reading file: /hello.txt
- read from file:
Hello World!
Renaming file /hello.txt to /foo.txt
- file renamed
Reading file: /foo.txt
- read from file:
Hello World!
Deleting file: /foo.txt
- file deleted
Testing file I/O with /test.txt
- writing................................................................
 - 1048576 bytes written in 10205 ms
- reading................................................................
- 1048576 bytes read in 522 ms
Deleting file: /test.txt
- file deleted
Test complete

`
```

lorol, 2020-09-25:
What happens if you upload the data partition again after the reboot/fix? Is it writable or it needs another reboot?
Check with other mklittlefs.exe versions.
Check with modified partition csv file that fills all 16MB to the end, Example, extend the last partition.

ramidomalys, 2020-09-25:
hello , yes i am sure , it is using mklittlefs.exe , because i can read the image perfectly ( after a hw reboot) . 
but the thing is , suppose that i dont want to flash a partition image , i want a clean partition and format it and then store some stuff .... 

apparently , flashing a new sketch reproduce the problem easily: 
this is how to produce the problem : 
- erase flash  / (or not)
- flash new sketch 
- flash image  ( or not ) 
-run : PROBLEM ( if no image is flashed it cant format anyway ) 
- HW restart 
- run : NO problem 
-re-flash exact same sketch 
- run : no problem 
-re-flash exact same sketch but with modifying the code : ( add simple delay in first line ) 
- run : PROBLEM
- Re-flash the partition image / HW Reset 
- run : NO problem
---------------------------------------------------------
uploading same/new image after HW reboot/fix without updating the sketch -> No problem
--------------------------------------------------------

ramidomalys, 2020-09-25:
this is the log after erase flash and flash only the sketch : 
```

rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
�.pio\libdeps\esp32\LittleFS_esp32\src\lfs.c:998:error: Corrupted dir pair at {0x0, 0x1}
E (2559) esp_littlefs: mount failed,  (-84)
E (2559) esp_littlefs: Failed to initialize LittleFS
.pio\libdeps\esp32\LittleFS_esp32\src\lfs.c:1685:debug: Bad block at 0x0
.pio\libdeps\esp32\LittleFS_esp32\src\lfs.c:1691:warn: Superblock 0x0 has become unwritable
E (2578) esp_littlefs: Failed to format filesystem
[E][LITTLEFS.cpp:81] format(): Formatting LITTLEFS failed! Error: -1
[E][LITTLEFS.cpp:56] begin(): Mounting LITTLEFS failed! Error: -1
```

lorol, 2020-09-25:
Can you please use IDE VS Code with PlatformIO extension and sample project to confirm same issue. 
I cannot reproduce it.

ramidomalys, 2020-09-30:
Hello again , sorry I was away from work
Yes I confirme to you, I am on vs code , with platformio  extension already. That's how I tested your example first place , 
I tried same exact project on a WROOM board ( 4MB) FLASH . I couldn't reproduce it.
Have you got a wrover with 16MB to reproduce same behavior ? It's only place where I am facing this issue ..thanks .

lorol, 2020-10-01:
Hi,
I don't have 16MB module to test. Please post your partition CSV for reference - someone else can test..
You also can post your issue on esp32 core repo in regards of SPIFFS, as you said has similar results.
I don't think it is anything to do with LittleFS.

ramidomalys, 2020-10-05:
Hello again , i just received Wrover dev kit  with 8MB flash, 
i could not reproduce the same bug ... i guess its the 16MB driver/hardware incompatibility, i'll look further into some more testing  and then i'll post this issue to espressif via my supplier.
 thanks for your time i'll let you know if anything comes up

lorol, 2020-10-05:
Ok @ramidomalys.
Thank you for the info. I will close the case for now. Feel free to post back any new findings.
I will also update if anything pops up on my side.

closed 2020-10-05 by lorol