lorol/LITTLEFS #4: Directories support

fdistorted, 2020-10-09:
Thank you for implementing LIttleFS for Arduino esp32.
I am migrating from esp8266 to esp32
My working esp8266 code looks next:
```
uint8_t get_chunks_number(const char *dir_path)
{
    uint8_t result = 0;
    Dir root = LITTLEFS.openDir(dir_path);
    while (root.next())
    {
        result++;
    }
    return result;
}
```
Is there a way to go through folder?

lorol, 2020-10-09:
@fdistorted 

Yes, you can go through folders. See in the example how it is done, including the differences between esp32 and esp8266:
https://github.com/lorol/ESPAsyncWebServer/blob/master/src/SPIFFSEditor.cpp#L185

Also parsing folders is shown here:
https://github.com/lorol/LITTLEFS/blob/master/examples/LittleFS_test/LittleFS_test.ino#L11

fdistorted, 2020-10-09:
Oh. I it see now. thanks.

closed 2020-10-09 by fdistorted