lorol/LITTLEFS #5: Finally

ullix, 2020-10-15:
I struggled with SPIFFS, finding it to be "A Flash Memory Catastrophe" [https://forum.arduino.cc/index.php?topic=648987](ESP32 mit SPIFFS - eine Speicher Katastrophe), using FFat with some success but issues on power failures, but wanting LittleFS for its claimed Power-loss resilience.

Many, many thanks for making it available in the Arduino libs!

I repeated my earlier tests, and can say that at least in writing performance LittleFS beats the others. 

[image]

SPIFFS is simply not usable. Why would one uses it when FFat and now LittleFs are avaialble?

There does not seem to any reason to simulate behavior of this very poor FS, so why even bother thinking about it?

ullix, 2020-10-16:
Well, kudos to all developers; unfortunately most of this stuff is beyond my pay grade...

The "Power-loss resilience" is still something I wanted to test. So I took the program used for the graphs above, and increased saving rate 10fold to once every 100ms. Then I interrupted the ESP32-Pico by pulling the USB-power plug or hitting the reset button on the device. With a saving cycle - open,print,close - typically taking some 50...80 ms, there would be a very good chance of knocking the cpu out in a middle of saving.

I tried hard, but I have to admit that I failed miserably trying to corrupt the Internal-Flash filesystem.

Wonderful :-))

Though I don't need it now, I am wondering whether LittleFS can also be used on an external SD card?

lorol, 2020-10-16:
@ullix 
Thanks again to continue testing. Your robustness validation is a great feedback!
Regarding [SD card, I believe currently it uses only FAT (ff.h)](https://github.com/espressif/arduino-esp32/tree/master/libraries/SD), but maybe @lbernstone  or  @BrianPugh can determine if LittleFS is possible to be implemented for esp32 and or if makes sense to go be done :) at all.

ullix, 2020-10-17:
On LittleFS: how do you know that you have a corrupt file system or at least one corrupt file?

Would be great if it never happens, but I am not ready yet to believe that. 

On FFat I routinely and often checked and looked for a file with this file name:

`const char bad[]  = {'/', 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, '.', 0xFF, 0xFF, 0xFF, 0x00};`

When you find only a single of them, the FFat FS is corrupt and it needs formatting. 

But what to check on LittleFS???

lorol, 2020-10-17:
You can use provided means: https://github.com/lorol/LITTLEFS/blob/master/src/LITTLEFS.cpp#L50
Further, you can check in lower level  files when ESP_FAIL and  ESP_OK are returned.

See also on ARMmbed discussions, like; https://github.com/ARMmbed/littlefs/issues/68#issuecomment-403868992

ullix, 2020-10-18:
I found an error code `LFS_ERR_CORRUPT`, but I don't see how to read it from the Arduino code?

Also, the ARM discussion seems to show that corruption can happen, but will be seen only as a bad directory, suggesting that the individual files in that dir can be read, although they may be correupt and contain nonsense?

It would seem that an enhancement to code is valuable which allows an easy call to determine a "corrupt" situation!

lorol, 2020-10-18:
@ullix At Arduino, you just should enjoy the high-level programming and don't worry too much :) 
If something is wrong and detectable, it will fail on mounting (begin ...) stage and require format anyway.
Looking at corrupted directory pair internals, trying to recover, chasing the readable files etc. is not something that I would recommend to go deploy at your application level.  Maybe implement your own verification at high level, example, a checksum, redundancy with several files for given thing etc.

lorol, 2020-10-21:
I am closing this case as it is rather a discussion, not a problem. 
You can still comment after the closure if necessary.

closed 2020-10-21 by lorol

EL-EK, 2021-06-22:
Just wondering if work was done on implementing littlefs on esp32 and SD card as discussed by @ullix ?  Are there thoughts from @BrianPugh or @lorol ?  I'd like to use littleds on an SD card with ~32GB of storage.