lorol/LITTLEFS #6: LITTLEFS(fs_info); causes error.

PQCraft, 2020-11-01:
```
#include <Arduino.h>
#include <FS.h>
#include <LITTLEFS.h>
#define  fs LITTLEFS
#include <stdlib.h>
struct FSInfo {
    size_t totalBytes;
    size_t usedBytes;
    size_t blockSize;
    size_t pageSize;
    size_t maxOpenFiles;
    size_t maxPathLength;
};
void setup() {
  FSInfo fs_info;
  LITTLEFS(fs_info);
}
```
This causes `no match for call to '(fs::LITTLEFSFS) (FSInfo&)'`
and when doing it the way it is supposed to be:
```
#include <Arduino.h>
#include <FS.h>
#include <LITTLEFS.h>
#define  fs LITTLEFS
#include <stdlib.h>
struct FSInfo {
    size_t totalBytes;
    size_t usedBytes;
    size_t blockSize;
    size_t pageSize;
    size_t maxOpenFiles;
    size_t maxPathLength;
};
void setup() {
  FSInfo fs_info;
  LittleFS.info(fs_info);
}
```
This causes `'LittleFS' was not declared in this scope`
I have even tried using FS.info and I get `expected unqualified-id before '.' token`
There is absolutely no reason any of these should give any errors at all. This should just simply work. I am very close to just giving up on making this stupid os.

PQCraft, 2020-11-01:
Also, I'm apparently THE ONLY PERSON IN THE WHOLE WIDE WORLD that has this problem and this is probably going to get acknowledged about 2 months after this was made, after I moved on to doing something else or used a different filesystem.

PQCraft, 2020-11-01:
As of right now, I have no way to see the free space on the disk which is crucial for what I'm doing. This means that everything is halted until this is fixed. The second method should've just worked fine but it doesn't for some reason. LittleFS is LittleFS. IDK why there are 914538143751758 different LittleFS command combinations and every tutorial uses a different method for EVERYTHING. Why in the F isn't this standardized?

PQCraft, 2020-11-01:
I am going prematurely bald bc of this. This is dumb.

PQCraft, 2020-11-01:
LITTLEFS.totalBytes() returns 1073479984 and LITTLEFS.usedBytes returns 1061162748. I know that the absolute max it can be is ~2000000. I am pretty sure I don't have a gig of flash on an ESP32.

PQCraft, 2020-11-01:
I also still cannot retrieve the volume label.

PQCraft, 2020-11-01:
It's returning those numbers on an unsigned int. wtf? the max is 65535.

PQCraft, 2020-11-01:
oh nvm. i was trying to grab the size before i used fs.begin(). Please disregard the last few messages except for the one about the volume label.

lorol, 2020-11-01:
Hi @PQCraft 
It will be better if you are less arrogant.

PQCraft, 2020-11-01:
Sorry. It was 2AM and my brain was melting. I was frustrated because every tutorial I saw on how to use this had a different method of doing everything and each method gave an error, or did not provide the information I needed (such as the volume label).

PQCraft, 2020-11-01:
@lorol Sorry. It was 2AM and my brain was melting. I was frustrated because every tutorial I saw on how to use this had a different method of doing everything and each method gave an error, or did not provide the information I needed (such as the volume label).

PQCraft, 2020-11-01:
@lorol Also, I kind of don't know what you took as arrogant.

lorol, 2020-11-01:
@PQCraft see here for the label, hope this helps:
 https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/spi_flash.html#api-reference-partition-table
 https://github.com/espressif/esp-idf/blob/master/examples/storage/partition_api/partition_find
example:
```
 #include "esp_partition.h" 
...                      
 const esp_partition_t * part  = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_FAT, NULL); 
    Serial.printf("\nFound partition '%s' at offset 0x%x with size 0x%x", part->label, part->address, part->size); 
```
Not sure why you need to dig into the label ... anyway. 
I am closing the ticket. 
Thank you.

closed 2020-11-01 by lorol