lorol/LITTLEFS #7: Recursive delete of directory  does not work as expected on LittleFS

gitpeut, 2020-11-04:
[Deltree.ino.txt](https://github.com/lorol/LITTLEFS/files/5485253/Deltree.ino.txt)

The attached Arduino sketch compiled for ESP32 fails to recursively delete a directory. When executed on an SD_MMC FAT filesystem the directory is succesfully removed.
When built with debug the following messages are shown :

/testdir/sub1 is a directory
E (1507) esp_littlefs: Failed to unlink path "/testdir/sub1/vier.txt". Has open FD.
result of removing file /testdir/sub1/vier.txt: 0
E (1539) esp_littlefs: Failed to unlink path "/testdir/sub1/vijf.txt". Has open FD.
result of removing file /testdir/sub1/vijf.txt: 0
E (1570) esp_littlefs: Failed to unlink path "/testdir/sub1/zes.txt". Has open FD.
result of removing file /testdir/sub1/zes.txt: 0
result of removing directory /testdir/sub1: 0

Is this a bug? Should I use another method to remove a directory tree?

lorol, 2020-11-04:
Please see this:
https://github.com/lorol/LITTLEFS/blob/master/examples/LittleFS_test/LittleFS_test.ino#L162

gitpeut, 2020-11-04:
Thx, I found already that removing the files one-by-one, without opening the directories did work without issues, but I
was hoping that maybe I overlooked something, or stumbled on an unintended bug,  so I could keep using the (to me) obvious method.
To confirm:  opening the directory and using openNextFile() on LittleFS to delete files is simply not possible/supported?
Is this the case forever or is this a feature that will be implemented in a future release?

lorol, 2020-11-04:
@BrianPugh 
Hi Brian, can you help in this?  it fires here: https://github.com/joltwallet/esp_littlefs/blob/master/src/esp_littlefs.c#L1260

gitpeut, 2020-11-04:
:-) no, closing the entry causes the property "name" to be inaccessible, and the following:

[E][vfs_api.cpp:126] remove(): bad arguments
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400014fd  PS      : 0x00060130  A0      : 0x800e37c0  A1      : 0x3ffb1a60  
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffb1d80  
A10     : 0x00000003  A11     : 0x00060723  A12     : 0x00060720  A13     : 0x00000001  
A14     : 0x00060723  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400014fd:0x3ffb1a60 0x400e37bd:0x3ffb1a70 0x400ead12:0x3ffb1d80 0x400ead4a:0x3ffb1e10 0x400d88d2:0x3ffb1e50 0x400d1799:0x3ffb1ef0 0x400d1895:0x3ffb1f70 0x400d8f9f:0x3ffb1fb0 0x4008975d:0x3ffb1fd0

I made a workaround deltree2 buffereing all filenames. Although it works I am not overjoyed by all the overhead, see attachment.
[Deltree2.ino.txt](https://github.com/lorol/LITTLEFS/files/5490542/Deltree2.ino.txt)

lorol, 2020-11-04:
There may be something to be worked on, But I cannot help you much on it, 
Eventually see esp_littlefs or at Arduino FS implementation your findings can be reviewed.
SPIFFS was first to ESP, then LittleFS ...

gitpeut, 2020-11-04:
After a bit of pondering I expanded on your idea:

```
if ( entry.isDirectory()  ){
         deltree( ff, entry.name() );        
      } else{ 
         char* tmpname = strdup( entry.name() ); // buffer file name
         entry.close();
         Serial.printf( "result of removing file %s: %d\n", tmpname, ff->remove( tmpname ) );
         free( tmpname );
      }     
```
I implemented the obvious :-). And this actually works on both LittleFS and the SD_MMC Fat file system.

I dug a bit deeper and I found that  the openNextFile goes down to the vfs_api.cpp  where it returns a shared pointer 
to a file. :

```
FileImplPtr VFSFileImpl::openNextFile(const char* mode)
...
return std::make_shared<VFSFileImpl>(_fs, name.c_str(), mode);

```
Maybe LittleFS is just more sophisticated than Fat in it's refusal to delete an open file.

Anyway, this workaround/solution  is perfectly acceptable to me.

BrianPugh, 2020-11-04:
Hey guys, so whats the current thinking here? Bug in our port of littlefs, or bug in esp-idf's virtual filesystem? The VFS is pretty good, but it does have it's issues around absolute POSIX compatability. I try to adhere to POSIX standards as closely as possible.

@gitpeut your latest code snippet for deleting a directory is how I would do it.

gitpeut, 2020-11-05:
Not sure. The arcane language at https://pubs.opengroup.org/onlinepubs/009695399/functions/readdir.html
does not say anything about opening the actual file inside the directory, where I get the impression ( and I might be very wrong, this is beyond my skills) that VFS actually opens this file for read. 
If that is the case, it's an esp-idf or a documentation issue, and LittleFS acts accordingly and correctly, and closing the file before deleting it is obligatory.

gitpeut, 2020-11-05:
Come to think of it, openNextFile is Arduino, not necessarily esp-idf. Arduino does not document if the file in the directory is opened or not.

gitpeut, 2020-11-05:
I propose to close this issue as "works as designed". I have no concerns about LittleFS left. Ok?

lorol, 2020-11-05:
Ok, I am closing it. 
Your "forced recursive folder deletion" solution is valuable!
Thank you.

closed 2020-11-05 by lorol