lorol/LITTLEFS #8: 'Dir' was not declared in this scope

PQCraft, 2020-11-08:
```
#include <Arduino.h>
#include <FS.h>
#include <LITTLEFS.h>
#include <esp_int_wdt.h>
#include <esp_task_wdt.h>
#include <esp32-hal-cpu.h>
#include <stdlib.h>
void setup() {
  Dir dir = fs.open("/");
}
```
The Arduino compiler throws `'Dir' was not declared in this scope`. Is there any other way I can list all the files in a directory?

PQCraft, 2020-11-08:
`LITTLEFS::Dir dir = fs.open("/");` causes `'LITTLEFS' is not a class, namespace, or enumeration`
and
`fs::Dir dir = fs.open("/");` does the same

lorol, 2020-11-08:
Use File, see here how:
https://github.com/lorol/ESPAsyncWebServer/blob/master/src/SPIFFSEditor.cpp#L187

PQCraft, 2020-11-08:
@lorol 
```
#include <Arduino.h>
#include <FS.h>
#include <LITTLEFS.h>
#include <esp_int_wdt.h>
#include <esp_task_wdt.h>
#include <esp32-hal-cpu.h>
#include <stdlib.h>
void setup() {
  String tmpStr = "";
  File dir = lfs.open("/");
  while (lfs.next()) {
    tmpStr += dir.fileName();
    tmpStr += " / ";
    tmpStr += dir.fileSize();
    tmpStr += "\r\n";
  }
  Serial.print(str);
}
```
`'class fs::LITTLEFSFS' has no member named 'next'`

lorol, 2020-11-08:
Please use openNextFile(); 
Everything is shown in the link in previous message.

PQCraft, 2020-11-08:
@lorol Ok thanks! It works now!

closed 2020-11-08 by PQCraft