lorol/LITTLEFS #9: LittleFS.usedBytes() reports twice the used bytes

everslick, 2020-11-08:
For me usedBytes() is off by a factor of 2.

[image]

This does not happen SPIFFS, FFAT nor SDFS. Anybody seen this too?

lorol, 2020-11-09:
Hi, can you please post the code to see when you report the total bytes?

everslick, 2020-11-09:
```
bool fs_usage(int64_t &total, int64_t &used, int64_t &unused) {
  total = used = unused = -1;

  if (!fs_mounted()) return (false);

#ifdef ESP32

  if (fs_on_sd) {
#ifdef HAVE_SDCARD
    total = SD.totalBytes();
    used  = SD.usedBytes();
#endif // HAVE_SDCARD
  } else {
#ifdef HAVE_LITTLEFS
    total = LittleFS.totalBytes();
    used  = LittleFS.usedBytes() / 2; // FIXME bug in LittleFS?
#endif // HAVE_LITTLEFS

#ifdef HAVE_SPIFFS
    total = SPIFFS.totalBytes();
    used  = SPIFFS.usedBytes();
#endif // HAVE_SPIFFS

#ifdef HAVE_FFATFS
    total = FFat.totalBytes();
    used  = FFat.totalBytes() - FFat.freeBytes();
#endif // HAVE_FFATFS
  }

#endif // ESP32

#ifdef ESP8266

  FSInfo info;

  rootfs->info(info);

#ifdef HAVE_LITTLEFS
  total = info.totalBytes;
  used  = info.usedBytes;
#endif // HAVE_LITTLEFS

#ifdef HAVE_SPIFFS
  total = info.totalBytes;
  used  = info.usedBytes;
#endif // HAVE_SPIFFS

#endif // ESP8266

  unused = total - used;

  if (unused < 0) unused = 0;

  return (true);
}
```

lorol, 2020-11-09:
Hi, the value is coming from the low level [esp_littlefs](https://github.com/joltwallet/esp_littlefs/blob/master/src/esp_littlefs.c#L133) and [lfs](https://github.com/littlefs-project/littlefs/blob/master/lfs.h#L620).
We probably need @BrianPugh to spend some precious time to check and report.

BrianPugh, 2020-11-09:
I just added an explicit test for this, so far I'm not seeing an issue on my end. I'll look through the above code now

https://github.com/joltwallet/esp_littlefs/commit/8adcce20d73d0a8184fd134735699992c6d963c3

lorol, 2020-11-09:
Thank you for looking to it. It could be something like integer types or when a path / folder is "open" to keep engaged once again. Maybe we need the whole sketch?

lorol, 2020-11-09:
@everslick I cannot reproduce your results. At least I don't have HW with >4M flash. What board and partition table you are using?

Aircoookie, 2020-11-10:
Cannot reproduce it either, added a 502 kB file to my FS and the reported used space increased by 508 kB (the extra kb are probably padding and an extra metadata block)

@everslick do you have files open and having been written to without calling file.close() yet? I believe I read in the upstream LFS repo somewhere that those temporary changes increase the `usedBytes` and calling close will return it to something more expected, so @lorol 's assumption is probably what's happening here.

everslick, 2020-11-12:
@Aircoookie Yes, the file is kept open all the time while I write one data record each second. Every 5 minutes I call flush() on the file, to make sure data loss in case of power failure is not too drastic. As soon as I close the file, numbers are reasonable again.

Having said that, I just double checked with LittleFS on ESP8266 and there also the blocks of an open file are counted twice. So this seems to be an LFS issue not an LFS@ESP32 issue. I will close this here. Thanks for participating to everyone and for porting LIttleFS to ESP32!

closed 2020-11-12 by everslick

lorol, 2020-11-12:
Thank you @everslick.
Just note the limitation of overwriting a large file when partition has no space to accommodate old + new content at same time.
If it is just appending, then no worries. LittleFS keeps the previous healthy data of opened file.