lorol/LITTLEFS #10: Major Writing Problem with LittleFS

ullix, 2020-11-11:
I reported before excellent performance with LittleFS. This was when I grew the file size by APPENDING new records. Now I attempted to OVERWRITE early records of a large file - like one would do in a circular buffer - and ran into major problems. I used the same writing method (file.print(text); with text being an exactly 100 bytes record of only ASCII chars ending with LF ) but now using the file open command with "rb+" instead of "a" and positioning the pointer with file.seek(position). But now the writing time exploded up to **11 SECONDS** for a 100 bytes record! This makes LittleFS useless for such an application. FFat does work.

I compared FFat, SPIFFS, and LittleFS. First graph repeats the previously reported APPENDING situation:

[image]

SPIFFS has 2 problems:
- you cannot fill the FS by more than 67%, no matter how long you try writing
- after a fill level of about 50% is reached, writing times jump to 2.5 sec per 100 bytes record

LittleFS and FFat
- both allow to fill the FS 100%
- writing times per records remain the same throughout fill level. LittleFS has a little bit of advantage, ans its max times are only 0.22 sec, while FFat has 0.34 sec.

Next test was to fill the file system with approx 45% with a file created in the append mode. The I began overwriting beginning at record 0. Graph shows results:

[image]

SPIFFS shows basically same behavior as before: filling limited to 2/3rd, write times jump to 2.5sec at approx. 50% fill level, but overwriting initial 45% is ok.

FFat also  behaves basically the same as before: 100% fill possible, writing times as before, except that during overwrite the average writing times are even a little bit shorter.

But LittleFS presents a huge problem! The writing times for the first record reach 11 SECONDS for 100 bytes! They slowly decrease down to the values seen with appending, but only when the 45% are overwritten, and every new record is basically being appended.

LittleFS cannot be used in any overwriting application, but FFat can. Unfortunately, during the course of testing I experienced several corrupted file systems when using FFat, but not with LittleFS. SPIFFS is no serious competition.  :-((

ullix, 2020-11-11:
I attach the code used to create the data above. I use the UECIDE IDE, not Arduino, but I verified that it compiles under Arduino too. Though make sure to read the important HOWTO notes in the top of the main file! (source code additions to FFat necessary when using Arduino; observe partitions settings!)

[BUGDEMO_FileSystems.zip](https://github.com/lorol/LITTLEFS/files/5523726/BUGDEMO_FileSystems.zip)

lorol, 2020-11-11:
@ullix Thanks again for the test results and comparison. 
For sure the robustness has a price.
Maybe (again) :) @BrianPugh is the best person to determine if this is expected by design/implementation or it can be improved.

lorol, 2020-11-11:
@ullix  Is it possible to tweak the littlefs filesystem parameters - like decrease the cache size and run again?
Also if you can observe if esp8266 implementation is better in this regard?

ullix, 2020-11-11:
@lorol Sure, the code is easy, take a look. But how do I tweak parameters? Can't do 8266, sorry.

ullix, 2020-11-11:
More observations. I uploaded a big file generated in the append mode, and then started overwriting this file always beginning at record 0 (zero) and measured the total time to write the record.

Then I shrunk the big file, uploaded and measured again, and repeated. This is the result:

[image]

The writing times increase linearly with the size to be overwritten. However, when touching a fill level of 50%, nothing is written anymore to the file, yet the "writing" does take time. There are no error messages. Testing file sizes after file.close() does not help, because in overwriting the file size does not change and the file.print() command always returns the expected bytes saved. What to do?

I don't know how LittleFS works, but from the data it looks like it has to shovel around all bytes above the position (= 0) where it wants to write. Therefore it fails completely when there are more than 50% to move, because there obviously is no space to make a duplicate.

If I am guessing right, this would require some major redesigning!

lorol, 2020-11-11:
> 
> 
> @lorol Sure, the code is easy, take a look. But how do I tweak parameters? Can't do 8266, sorry.
For the tests above you can decrease say down to 64 one by one in esp_littlefs.c:
```
#define CONFIG_LITTLEFS_WRITE_SIZE 128
#define CONFIG_LITTLEFS_LOOKAHEAD_SIZE 128
#define CONFIG_LITTLEFS_CACHE_SIZE 512
```
And in general, the overwriting you try would be troubled by the caching mechanism.

ullix, 2020-11-11:
For completeness: no problems with FFat

[image]

ullix, 2020-11-11:
When changing this line `#define CONFIG_LITTLEFS_CACHE_SIZE 512` to cache size 64 the ESP32 crashes with this report `"abort() was called at PC 0x400e522f on core 1"` when executing `myFS->begin(FORMAT_ON_FAIL);`.

When changing to cache 128, it works, but has no benefit:

[image]

I don't think that any tweaking will help; there is a fundamental issue behind it!

BrianPugh, 2020-11-11:
>The writing times increase linearly with the size to be overwritten. However, when touching a fill level of 50%, nothing is written anymore to the file, yet the "writing" does take time. There are no error messages.

This is concerning. To be clear, to replicate this, is it the following steps?
1. Create a file >50% of the file system
2. Start prepending data.

As for prepend/insert performance, this is a [design gamble the upstream littlefs authors made](https://github.com/littlefs-project/littlefs/blob/master/DESIGN.md#ctz-skip-lists). Most important bits quoted below:
>  If we update a block in the middle, we still need to copy the following blocks, but can reuse any blocks before it. Since most file writes are linear, this design gambles that appends are the most common type of data update.

I'll look at your code in a bit to see if this design issue is being exacerbated in a worst-case-situation type deal.

lorol, 2020-11-11:
Interesting, I just did a quick test - saved a 900kb text file to 1.5M LITTLEFS partition together with other smaller files.
If I try to modify (change/append/remove) and overwrite about same size to same file name - it fails and retains the exiting one.
However if I overwrite same file but with low size contents, say only 10k - it works.
I guess it really requires the already taken space + the new data to be able to fit to total bytes + some more,  regardless if the new data is intended to overwrite fully or partially the existing one.

lorol, 2020-11-11:
Tried same as above with Ffat - it allows to modify and overwrite 900kb on 1.5M fat partition!
Tried with SPIFFS but it crashed and application rebooted many times, giving up.

lorol, 2020-11-11:
Update: 
My statement 2 messages above is not exact.
if LITTLEFS file is open in pure appending mode like
```
fs::File f = _fs.open(filename, "a");
```
and you append more data it actually allows overwriting (updating) of file that is >50% of the partition.

ullix, 2020-11-12:
@BrianPugh "Prepending" is not the correct description. I use overwriting of existing data.

The idea is a circular buffer for data logging. Let's say the file is allowed to grow up to 1000 records. The 1001st record would then be written at position 0, overwriting the original first record. The file would then always have the latest 1000 records, and only those, and never get any bigger. (In my real application I plan to use two buffers, together occupying 90% of the Flash space)

Isn't your quote already the explanation for the behavior:

> If we update a block in the middle, we still need to copy the following blocks, but can reuse any blocks before it. Since most file writes are linear, this design gambles that appends are the most common type of data update.

If the file were 60%, and I attempted to overwrite its first record at position 0, then there is nothing to reuse, and all 60% needs to be copied. But since there is only 40% left, it has to fail! That is consistent with my data. No?

@lorol You seem to be confirming the very same thing.

But note that in "appending" mode I found the LittleFs performing very well, see the middle pic of the first graph in this thread, and  filled the FS 100%. It even tended to be a little bit faster than FFat. This appending behavior (combined with the blackout resilience) was what got me so excited about LittleFS. But this also seems to be consistent with the observed behavior: when you "append", then you write at the last position, and there is nothing that needs to be moved. 

But the overwriting problem is a real bummer. Is it really necessary in the concept of LittleFS that all the blocks after the writing position have to be moved? Right now it seems to be the case, but why can't moving a single block not be enough?

lorol, 2020-11-12:
@ullix 
The test code sketch you provided and all you findings are appreciated! (I just had to change <Time.h> to <TimeLib.h> to run it on Arduino IDE)

As a recap, LittleFS is the best choice except you need to overwrite a file with near or larger new data than the free space in the partition and the error response comes from lower level lfs after you start such attempt. The result is silent (for your app) fall-safe keeping of old data. 
By appending ("a" mode) of even a large file, the remaining free space of partition can be filled.

I believe, for this issue nothing further can be done, it looks a design limitation.
We can keep it opened for discussion, no problem.

@ullix - I would only recommend to reformulate somehow the title because it sounds a bit "apocalyptic"  :)

BrianPugh, 2020-11-12:
> CTZ skip-lists give us a COW data structure that is easily traversable in O(n), can be appended in O(1), and can be read in O(n log n)

Glossing over a bit the overwriting costs.

This issue is probably, at its core, the same issue you're seeing:
https://github.com/littlefs-project/littlefs/issues/451

Looking more into lfs's design, overwriting at the beginning of a file is pretty much worst-case-scenario.

@ullix I'm willing to bet you can get *much better performance* by chunking up your log into a few separate files, and cycling through them where upon a log file reaching a certain size you:

1. Select the next file
2. Delete the file
3. Start writing a new file, so all writes are appending.

It seems like besides that, this is just a fundamental design tradeoff made by littlefs and there's nothing we can do.

@lorol do we have a bug with overwriting in the middle of large files? Nothing should ever fail silently/without an error code.

lorol, 2020-11-12:
***Do we have a bug with overwriting in the middle of large files? Nothing should ever fail silently/without an error code?***
@BrianPugh I think we do ... The best way is if you include a test, say keep overwriting at >50% scenario. 
We had some early case see https://github.com/lorol/LITTLEFS/issues/1 wit difficulties to catch at high level. I know the lfs returns the issue but maybe is "too late" for higher wrapper or I am wrong?

BrianPugh, 2020-11-12:
I'll write some explicit tests; if we can fix it/report an error, the great. If we cannot, then at least we'll document it.

ullix, 2020-11-13:
@BrianPugh is it really impossible to modify LittleFS in such a way that instead of moving the whole file above the position, only the block is moved, into which I want to write? And the rest is obtained by juggling the pointers to the blocks? May be naive, sorry, but I really would like to use your FS!

lorol, 2020-11-13:
@ullix 
For your proposal you can open a case at https://github.com/littlefs-project/littlefs  to see if it fits to the robustest criteria.
BTW, if you delete the big file before rewriting, you can fill "more space". But you can lose the file in case of power failure.

ullix, 2020-11-13:
@BrianPugh 

> But you can lose the file in case of power failure.

Then why use LittleFS at all? No, the resilience to power failures is what I consider the one key component of LittleFS. Writing and overwriting is done by FFat quite well, as you see in my graphs.

lorol, 2020-11-13:
> Then why use LittleFS at all? No, the resilience to power failures is what I consider the one key component of LittleFS.

@ullix  Let see ... but that's why your proposal cannot be implemented and keep resilience to power at same time.  Isn't something like this in FAT?

BrianPugh, 2020-11-13:
@ullix if you choose to use littlefs, probably the solution that will get you to a working answer quickest is to have your logs circulate around a few files. This obtains approximately what you want, maintains high performance and strong data consistency, and doesn't require intervention from upstream littlfs devs.

ullix, 2020-11-13:
@lorol I have never heard a claim that FFat is power failure resilient? But if there were one, I can certify to the contrary :-(.
But I also got a corrupted LittleFS; no idea where this came from, might have to do with a bug, not so much with power failure. A little bit more error reporting by LittleFS would be really, really welcome!

@BrianPugh Let's do some calcs: I currently use two log files, one is 500k, the other is 800k, together 1.3MB. The file writing is blocking, so as I need some action every second, the saving must be completed within say 500 millisec. Taking the writing times I reported above, this means my files cannot be bigger than 25k (-> approx 500ms when overwriting pos 0). So I'd need to juggle some 50+ files.

Hmmm. What is worse: doing those 50+ files or looking into the LittleFS code? I guess it is a close call ;-)

BrianPugh, 2020-11-13:
I mean, changing a fundamental design decision in a filesystem sounds pretty challenging.

BrianPugh, 2020-11-13:
also it doesn't matter the number of files, the implementation would be the same, whether it be 2 log files or 50.

ullix, 2020-11-13:
@BrianPugh I see with delight that you are already thinking about it! According to @lorol a few people were involved in the making of LittleFS. So what kind of bounty would one have to set to get a BttrFS?

lorol, 2020-11-13:
@ullix I didn't mean FAT is power failure resilient, rather has a file chunks moving similar to one you asked for (I believe).
:)  Feel free to develop AnyNewFS and bring it here ...

ullix, 2020-11-14:
Finding a new name may be a challenge even I could master ;-)

FFat is working just fine, be it appending or be it overwriting. "If" it works. The "else" part to the "if" is the problem.

There is a reason why there is an Eject command for FAT formatted USB-sticks. Perhaps we need a UPS for microchips?

lorol, 2020-11-14:
@BrianPugh 
https://github.com/esp8266/Arduino/commit/c663c55926f205723c3d56dd7030bacbe7960f8e
Please see this. I did some tests on esp8266 and indeed it allows to use the space on overwriting!
Not sure if this actually deletes/invalidates the returning state and contradicts to the concept but it works. 
(on high level it looks like it is been deleted to free the space the overwritten with new content fine)

Regarding the error transparency from low to high level, maybe you can find ideas in esp8266 implementation. 
I will test few things to confirm if fails silently there on corner cases of overfill.

lorol, 2020-11-14:
@ullix see https://github.com/littlefs-project/littlefs/issues/244 
They mention B+-tree :)

ullix, 2020-11-15:
:-) I am hopeful. FFat file corruption keeps bugging me :-(

lorol, 2020-11-15:
@ullix 
Can you please apply the patch above and redo your test to the end of Littlefs partition?

ullix, 2020-11-15:
Sorry, but I don't know how to apply the patch. Can you provide me with a file and tell me where to copy it to? 

Which test are you looking for: appending or overwriting?

BrianPugh, 2020-11-15:
you can hold on for a moment, i'm almost done applying this to master

BrianPugh, 2020-11-15:
oh, I forgot we're on @lorol repo ;)

I added the sync-on-open fix and the corresponding unit test on this branch:
https://github.com/joltwallet/esp_littlefs/tree/large-file

I didn't notice any "silent" errors on the unit test before the fix was inplace. When writing, if there were no free blocks left, commands like `fwrite` would return 0 instead of the expected `n_memb*size`, which indicates that no bytes were written (and should be interpreted as an error).

lorol, 2020-11-15:
@BrianPugh 

I see. So we go with this. I will apply it manually here, too,  once you finalize.
Would this help also on you other open case (for duplicate opening etc.)?
As I am understanding the things (correct me if I am wrong), this is very similar to explicitly erasing the exiting file before rewriting it with new content. If a power failure occurs before file is closed etc. then it will have 0 bytes.

My plan is to wait an esp32 core release 1.x.x  (for IDF 3.3) and release a compatible LITTLEFS release here.
And for next esp32 core 2.x.x - it will be an internal library and no need of manual follow-ups.

BrianPugh, 2020-11-15:
Correct, this is very similar to deleting the file, then writing. So in the event of power-loss before the new content is flushed/synced, the file will be 0 bytes. However, I believe that this is expected operation whenever you open an existing file in write mode.

BrianPugh, 2020-11-15:
Also doesn't really help on the other multiple-file-descriptors-for-the-same-file issue, they're fairly unrelated.

lorol, 2020-11-15:
Great. As long as it is still corruption-proof, I believe price of loosing the previous copy is worth the ability to fill the free space, as it happens with the other FSs

lorol, 2020-11-16:
@ullix 
Please test with latest update,

closed 2020-11-16 by lorol

ullix, 2020-11-16:
it does not compile, neither in Arduino nor in UECIDE. Same error, here the message from Arduino:

        .../LITTLEFS-master/src/esp_littlefs.c: In function 'esp_vfs_littlefs_register':
        .../LITTLEFS-master/src/esp_littlefs.c:203:9: error: unknown field 'utime_p' specified in initializer
             .utime_p     = &vfs_littlefs_utime,
             ^

I got the zipped code from https://github.com/lorol/LITTLEFS.

lorol, 2020-11-16:
@ullix 
You need to un-comment #define CONFIG_LITTLEFS_FOR_IDF_3_2 
I am leaving it commented because the next esp32 core release 1.x.x will be on IDF3.3 or later.

ullix, 2020-11-16:
ok, got it running. But I am wondering whether I got the right lib, because all I can say is it is unchanged? The lib properties say "name=LittleFS_esp32, version=1.0.5, author=lorol" and I found the only other version info in lfs.h as "Software Lib  'LFS_VERSION'  : 2.2, On Disk Data Struct 'LFS_DISK_VERSION': 2.0". I suppose it is right?

I watched for errors reported by file.open, file.seek, and file.write: there never were any. I had the system debug level set to verbose: plenty of blabbing by the web server, but never a single message from the file system.

Again I used records of 100 bytes, all ASCII, and filled the FS completely by running in the append mode. It allowed 6+ recs/sec including all the overhead by my program, quite nice! Final fill level was 99.7%; only a single 4k block remained free. Writing seemed to continue at the normal speed, but of course nothing was written. No error of any kind.

[image]

The graph shows 14921 records written in 40 min. Writing times per records 50 ... 260 ms. Nothing unusual, except that I drew a Moving average (yellow trace) which highlights the increase in writing times from low fill (~70ms) to high fill (~100ms). I'd expect a dependency of writing times on the fill level for a hard disk with rotating platters, but for a Flash memory? Is that expected?

Now that the FS was 99.7% full, I tried overwriting record 0 and record 7000 (~in the middle). Writing took some 10+ sec, no error, and nothing was ever written.

Then I reduced file size to 7000 recs, 47.3% fill level. Overwriting record 0 worked, but took 13 sec. Overwriting rec 6500 took about 1 sec, and after it passed the 7000 rec end of the file, it continued with the normal fast speed.

I don't see anything different in performance, and surely no error reporting. What was expected?

lorol, 2020-11-16:
Strange, when you tried to overwrite at full state w/o success - did the old file remain or it was wiped off to 0 size?
Please download the zip again. You should be able to write on less than file size available space.

lorol, 2020-11-16:
The zip from repo not release.

BrianPugh, 2020-11-16:
whenever you are writing, are you verifying the returned number of bytes written matches the expected number of bytes written?

See this section on writing performance as the filesystem fills up:
https://github.com/littlefs-project/littlefs/blob/master/DESIGN.md#metadata-pairs

Specifically this graph:
https://raw.githubusercontent.com/geky/littlefs/gh-images/metadata-cost.svg?sanitize=true

Its my understanding that as the filesystem fills up, the average metadata pair occupancy goes up since they can no longer be prematurely split, so writing costs going up.

lorol, 2020-11-16:
@BrianPugh  
A question to you. at esp_littlefs.c level, if you keep writing, wouldn't it use cache and is it possible to return that data is taken but actually you don't have more physical space to write to flash?
I am suspecting something like this is happening when file.write() is used on high level.

BrianPugh, 2020-11-16:
I can write a test later today that does the following:
1. Opens like 10 file descriptors
2. Round-robin writes a single byte to each, ensuring that a 1 is returned. Don't do any explicit flushing/syncing
3. Whenever bytes stop being written, compare the total number of bytes to the original amount of free space in the file system.

That should get us some answers.

lorol, 2020-11-16:
@BrianPugh 
Great. You can pre-fill with a large file and leave 50-100k free in partition for the test in question.
I am trying something similar.

lorol, 2020-11-17:
I ran an Arduino sketch to write 10000 bytes in many files.
Before starting, I upload an image with 900k file to prefill the partition.
I use  file.write()  or  file.print(), 10 bytes repeating cycle and write to Serial the result of these functions.
On all 3,  file.write()  it keeps reporting more or less that 10 bytes are written up to 10000 but in reality it is not true.
The problem cannot be caught at that level.
Only lfs returns an error "underneath" 

LITTLEFS
10000 bytes wrote to file. File size: 10000
...
    lfs.c:495:error: No more free space 595 (repeats on ~128 bytes try to write)
...
9926 bytes wrote to file.  File size: 0

SPIFFS
10000 bytes wrote to file. File size: 10000
...
9978 bytes wrote to file. File size: 1280

FFat
10000 bytes wrote to file. File size: 10000
...
10000 bytes wrote to file. File size: 0

BrianPugh, 2020-11-17:
@lorol so I wrote a test here:

https://github.com/joltwallet/esp_littlefs/commit/d70e2ad65bb1b1efbb7d9d65f75d822f667d3031

Basically it:
1. Writes and flushes a bulk file such that 100,000 bytes should remain according to `esp_littlefs_info`.
2. Opens 7 file descriptors.
3. Round robin write a byte to each file without explicitly flushing. Once a `fwrite` call returns 0 instead of 1, break. Record the number of bytes supposedly written by these `fwrite` calls.
4. Close all the files
5. Get the cumulative file size of these 7 files.

The important bits of the test are as follows:
```
File 3 fwrite returned 0
File 0 is 12288 long
File 1 is 12288 long
File 2 is 12288 long
File 3 is 0 long
File 4 is 12287
File 5 is 12287
File 6 is 12287
expected_bytes: 86012
actual_bytes:   73725
```

It seems like the file that returned `0` on `fwrite` just gets deleted? Need to dive into this a bit more.

ullix, 2020-11-17:
@lorol why would another download get any different answers - the versions haven't changed?

lorol, 2020-11-17:
@ullix the "version" is just a string, used for releases and tags. 
The important is the commit number; https://github.com/lorol/LITTLEFS/commits/master

@BrianPugh 
Can your result be exactly because on last change fsync actually makes a 0-size file upfront on opening for write then it could not fit to remaining space and normally nothing is written?

More concerning is on Arduino, because the file.write() or file.print() return incorrectly 10 of 10 successively written bytes, even you do file.flush() so no nice way to catch room limit. It looks like on other FS is similar.

lorol, 2020-11-17:
The three FS have unreliable file.write() or file.print() return values on low-space scenario.
However, on LITTLEFS they never return 0, which is a downside.
On FFat and SPIFFS at least if current file faces the issue and ends up with 0 or wrong size, 
an attempt to write another file, file.write() returns 0 which is somewhat usable. 
It will be good if LITTLEFS has similar output.

BrianPugh, 2020-11-17:
So if I `sync` periodically, that (of course) writes it to disk. I'm just surprised that somehow 12KB of data is floating around without some of it being committed to the file on disk.

So bright side, from your tests, I guess its not an lfs-specific issue. But, that still doesn't really provide a good environment for a dev. That said, I'm not sure what we can do to make this more fool-proof.

@lorol are you saying you never have `fwrite` (or equiv) return 0? In my tests they return 0 when the device is out of space, but it seems relatively catastrophic (a `fsync` afterwards doesn't save what was recorded so far).

lorol, 2020-11-17:
@BrianPugh 
I Just repeated several times the test. It eventually returns 0. but a bit "later". 
Weird ... some buffering or so cannot be seen immediately on surface app level.

10000: bytes wrote to file. File size: 110
lfs.c:495:error: No more free space 430
0: bytes wrote to file. File size: 0

If I change the file name sequences to not overwrite after reboot and flash again to continue:
10000: bytes wrote to file. File size: 150
lfs.c:495:error: No more free space 657
and then file.write() reports 0 on next file
0: bytes wrote to file. File size: 0

lorol, 2020-11-17:
It looks like only when space is very low, then  file.write() returns 0 and no new file is written.
Previous files are normal 10000 bytes long and keep attempting to write to new file 10 bytes and flush.
from tst_10.txt lfs errors start popping
After tst_18.txt, new files are attempted (no errors on opening or so) but file.write() returns 0 continuously and no real file saved to flash.
Here is the last part of directory list after I flash another sketch - just to list - and keep the partition from the previous test untouched.

  FILE: /tst_10.txt  SIZE: 4090  LAST WRITE: 1970-01-01 
  FILE: /tst_11.txt  SIZE: 520  LAST WRITE: 1970-01-01 01:01:13
  FILE: /tst_12.txt  SIZE: 510  LAST WRITE: 1970-01-01 01:01:59
  FILE: /tst_13.txt  SIZE: 510  LAST WRITE: 1970-01-01 01:02:34
  FILE: /tst_14.txt  SIZE: 510  LAST WRITE: 1970-01-01 01:03:16
  FILE: /tst_15.txt  SIZE: 510  LAST WRITE: 1970-01-01 01:03:59
  FILE: /tst_16.txt  SIZE: 510  LAST WRITE: 1970-01-01 01:04:38
  FILE: /tst_17.txt  SIZE: 510  LAST WRITE: 1970-01-01 01:05:25
  FILE: /tst_18.txt  SIZE: 120  LAST WRITE: 1970-01-01 01:06:15

lorol, 2020-11-18:
with lfs trace enabled:
 lfs.c:3006:trace: lfs_file_write -> 10
 lfs.c:2743:trace: lfs_file_sync(0x3ffb8580, 0x3ffb8d74)
 lfs.c:2795:trace: lfs_file_sync -> 0
 lfs.c:2882:trace: lfs_file_write(0x3ffb8580, 0x3ffb8d74, 0x3ffb8df0, 10)
 lfs.c:3006:trace: lfs_file_write -> 10
 lfs.c:2743:trace: lfs_file_sync(0x3ffb8580, 0x3ffb8d74)
 lfs.c:2795:trace: lfs_file_sync -> 0
 lfs.c:2882:trace: lfs_file_write(0x3ffb8580, 0x3ffb8d74, 0x3ffb8df0, 10)
 lfs.c:2930:trace: lfs_file_write -> -28
 lfs.c:2743:trace: lfs_file_sync(0x3ffb8580, 0x3ffb8d74)
 lfs.c:2748:trace: lfs_file_sync -> 0
 lfs.c:2882:trace: lfs_file_write(0x3ffb8580, 0x3ffb8d74, 0x3ffb8df0, 10)
 lfs.c:2930:trace: lfs_file_write -> -28
 lfs.c:2743:trace: lfs_file_sync(0x3ffb8580, 0x3ffb8d74)
 lfs.c:2748:trace: lfs_file_sync -> 0

trullock, 2021-12-28:
@ullix Could you help me with my own LittleFS profiling problem, how are you capturing the data for your graphs in your original post? Thanks

ullix, 2021-12-28:
@trullock In the 2nd post from top https://github.com/lorol/LITTLEFS/issues/10#issuecomment-725374559 you find a link to the code used on the ESP32.

The receiving software is my open-source package GeigerLog, download from https://sourceforge.net/projects/geigerlog/

I configured GeigerLog for the use of device "AmbioMon". Thus GeigerLog collected the data from the ESP32 via the ESP-generated website "http://<IP of ESP32>/lastdata" with a rate of once per second. 

This used the ESP function "handle_lastdata" to provide the data, which you find in the webSTA.ino file as the last function. For more, search all *.ino for the word "GeigerLog".

Any other software which can call the ESP regularly should also work. I guess this includes curl in a script to create a CSV data file. Then load this file into GeigerLog, or into a spreadsheet.

[1 more reply]