lorol/LITTLEFS #12: LittleFS function available for ESP8266 but not for ESP32

ullix, 2020-11-18:
Apparently there are a few things available for ESP8266 which are not accessible for the ESP32, according to https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html#littlefs-file-system-limitations, like info:

        struct FSInfo {
            size_t totalBytes;
            size_t usedBytes;
            size_t blockSize;
            size_t pageSize;
            size_t maxOpenFiles;
            size_t maxPathLength;
        };

        FSInfo fs_info;
        SPIFFS.info(fs_info);
        or LittleFS.info(fs_info);

At least not under Arduino. Is there a way to read e.g. info and other stuff for the ESP32?

lorol, 2020-11-18:
Hi, on esp32 it is another way to do this. It is similar for SPIFFS and FFat. 
.info is not available by design.
Unfortunately the full documentation is not done but you can see roughly the methods and equivalents here: 
https://diyprojects.io/esp32-get-started-spiff-library-read-write-modify-files/

Besides that, maxOpenFiles is not used in LITTLEFS, no such limitation to be defined. It is kept just as parameter to match open().
see what is actually available:
https://github.com/lorol/LITTLEFS/blob/master/src/LITTLEFS.h#L24

ullix, 2020-11-18:
Yes, so far the source code had been my only source of info. Docu would be really welcome!
What among other I am trying to figure out is how many files can be created on LittleFS? I ran a loop up to 1000, but that cannot be right, assuming that each file needs at least 1 block of 4096 bytes? I did not see any debug error messages.

lorol, 2020-11-18:
Hi, can you figure out how many files you can create on SPIFFS and FFat?
Unfortunately I cannot help with the debugging other than point you how to enable on lower level, until the library is in this hacky way. Once is moved built-in to the core it cannot be done even this way.
Again if you give me example with esp32 SPIFFS what from it is good to have, I can work with the devs to get it implemented if possible. But not from esp8266. It is different "animal"

ullix, 2020-11-19:
Google did not help getting the info, so I had to resort to testing. I present the (confusing) results, hopefully it will help others.

For this post:   1 Block == 4096 Bytes.

On my ESP32 with 4MB flash, the partitioning is this, applicable to both SPIFFS and LittleFS

        # Espressif ESP32 Partition Table
        # Name,  Type,   SubType,    Offset,     Size,      Flags
        nvs,     data,   nvs,        0x9000,     0x5000,
        otadata, data,   ota,        0xe000,     0x2000,
        app0,    app,    ota_0,      0x10000,    0x140000,
        app1,    app,    ota_1,      0x150000,   0x140000,
        #eeprom, data,   153,        0x290000,   0x1000,
        spiffs,  data,   spiffs,     0x290000,   0x170000,
        #ffat,    data,   fat,        0x290000,   0x170000,

For FFat, comment out line spiffs, and un-comment line ffat.

The partition size given to any file system, FFat, SPIFFS, or LittleFS, is always 0x170000 bytes, or 368 Blocks. Of this the FS take different amounts for themselves:

    LittleFS : total: 0x170000 Bytes, consumed by LittleFS: 2 Blocks, free: 366 Blocks
    FFat     : total: 0x164000 Bytes, consumed by FFat:     nothing,  free: 356 Blocks
    SPIFFS   : total: 0x1507c1 Bytes, consumed by SPIFFS:   nothing,  free: 336.485 Blocks (yes, fractional)

The net effect is that for the same sized partition LittleFS has 10 Blocks more than FFat, and 29.5 Blocks more than SPIFFS!

LittleFS is presented as a Block oriented FS, and in file `esp_littlefs.c` one finds: `#define CONFIG_LITTLEFS_BLOCK_SIZE 4096 /* ESP32 can only operate at 4kb */`. Looks like the Block is cast in concrete, however:

**FFat:** This is almost straight forward: One file = at least One Block. A file size of 4097 Bytes requires 2 Blocks. I could create only 306 files of 4096 bytes each, occupying exactly 306 Blocks. 50 Blocks remained free, but I could not create a single additional file in that space! 

**SPIFFS:** I could create 1000 files of 100 Bytes each, taking 122.56 Blocks, which amounts to 502 Bytes per 100 Byte file. The total is only 36% of fill level. As I know that big problems start at ~50% fill level (https://github.com/lorol/LITTLEFS/issues/10 ), making the FS unusable for me, I did not investigate further.

**LittleFS:** While the LittleFS is Block oriented, this is not the full truth. I could create 1000 files of 100 Bytes each, occupying a FS space of under 50% fill level! This amounts to ~720 Bytes per 100 Byte file. Reading them back worked for each file, and always showed proper content. File creation times went up from ~10 ms for the first dozen to 1 second for most, many peaking at **1.4 seconds**!

Eventually I figured out that with file sizes of 512 bytes or smaller, several files fit into one Block. From 513 bytes upwards, a full Block is needed. And beyond 4096 bytes an new block is needed for the 4097th Byte, and so on.

Going to file sizes of 4096 Bytes, I could create 352 files. But they did not occupy 352 Blocks, but 368, filling the whole FS space. Free Bytes were zero! So, 16 Blocks are missing, probably used for some kind of undefined overhead. File creation times were mostly in the sub-100 millisec, but several times they went up to above **2 seconds**!

Somewhat disconcerting was the fact that, despite debug messages being enabled, the file creation simply failed without error message, only that byteswritten = 0, and filesize= 0  after flush. 

And for completeness: with file size 8192 Bytes I could create 121 files, taking 367 Blocks ( 1 Block unused). This is an overhead of 125 Blocks!!! (So strange, I repeated with a freshly formatted file system, but got exact same result). No debug messages of any kind.

File creation times mostly around 100ms, but sometimes again popping up to over **2 seconds!**

I am mulling over some superstructure on top of LittleFS. Only files of 1 Block size seem to make sense?

lorol, 2020-11-19:
@ullix 
Thank you for the time spend to test.
Note FFat has one (first) block unused. The partition is practically w/ offset 1 block.
I also faced strange results with 368 blocks math when filling near partition capacity (and asked myself, how much n USB stick or HD can be really full?) 
So the practical takeaway for me is simply to prevent writing (if possible) much earlier than full capacity by application means, eg. check what's left somehow before attempt a new file writing.

lorol, 2020-11-19:
A bit off-topic too;  https://stackoverflow.com/questions/59817278/esp32-eeprom-read-write-cycle
Not sure if this is OK with your file logging requirement using the flash.
10k rw cycles is not too much

ullix, 2020-11-20:
Regarding wear leveling, a data logger - which I am building - is almost the perfect application: you start writing continuously from one end to other, and then overwrite from the beginning. Spreading the use of memory usage more evenly is not possible. The limit in write cycles is not necessarily the most pressing concern. Assuming that I keep data for at least 10 days, it follows that each byte is written every 10 days, and 10000 cycles would thus last 27 years. And the new EEPROM-like chips called FRAM allow 10E13(!) write cycles. That is 30000 years writing each byte 10 times per sec!

The available sizes in Bytes are more of a problem. Ideally I would use the whole 366 Blocks of ESP32 space now used by the file system for EEPROM. But the EEPROM use on an ESP32 is a simulation limited to 1 Block, and it is shadowed in RAM (to reduce write cycles). So, anything like 300 Blocks is currently out of the question.

External EEPROMs are currently available up to 4MBit, and FRAMs up to 2MBit, equiv to 128 and 64 Blocks, resp., i.e. requiring 3 and 6, resp, chips to install next to the ESP. Some chips have 2 address pins, limiting to only 4 chips, but some have 3, allowing up to 8 chips. Not impossible, but a bit of BOM and real-estate on the PCB problem.

A much bigger problem is the use of EEPROM/FRAM with the I2C interface, because the Wire lib conflicts with the ESPAsyncWebserver (https://github.com/me-no-dev/ESPAsyncWebServer/issues/710 ) resulting in crashes consistently. Choosing a simpler webserver doesn't help, as that one doesn't support loading multiple parts of a web page, like CSS, JS (https://github.com/espressif/arduino-esp32/issues/3890 ). I haven't tried chips with the SPI interface. 

So far stuck with the file system on an ESP32 flash. Still hoping for LittleFS.

lorol, 2020-11-24:
Closing as it gets off-topic.

closed 2020-11-24 by lorol

trullock, 2023-08-15:
@lorol your diyprojects.io url is dead, what was the actual answer to this question?