lorol/LITTLEFS #13: arduino-esp32fs-plugin upload fails...

mike-s123, 2020-11-23:
arduino-esp32fs-plugin doesn't allow posting issues, so it's here. esptool.py can fail uploading some images when compression is used. Might want to add **--no-compress** when calling it. Unfortunately, they don't offer a configurable timeout. More info [here](https://github.com/earlephilhower/mklittlefs/issues/11).

lorol, 2020-11-23:
Ok, will consider.

mike-s123, 2020-11-23:
To answer your question from the other page, it does add to the upload time. How much, I suppose, depends on how compressible the image is. A large image with few files probably can be compressed a lot. I think my ~13MB image, which has about 6.5MB of files in it, compressed to about 5.5MB. I did submit an enhancement request for esptool.py, to add a configurable timeout, which might also be a solution.

mike-s123, 2020-11-24:
@lorol Looks like they're going to [fix it upstream](https://github.com/espressif/esptool/issues/577).

closed 2020-11-24 by mike-s123