lorol/LITTLEFS #14: Error compiling library

AcuarioCat, 2020-11-25:
Hi,
when I compile the library I'm getting an error (I'm just trying one of the examples):
I'm using the latest version from Github.

`esp_littlefs.c: In function esp_vfs_littlefs_register
 
esp_littlefs.c: 203:9: error: unknown field 'utime_p' specified in initializer
   utime_p     = &vfs_littlefs_utime`

lorol, 2020-11-25:
Hi
Uncomment the line
 #define CONFIG_LITTLEFS_FOR_IDF_3_2 
in esp_littlefs.c.

closed 2020-11-25 by lorol

WoodyLetsCode, 2020-12-21:
What does "_old IDF_" mean? 
What do I have to do that I don't have to uncomment this line?

lorol, 2020-12-21:
Are you using platformio?

WoodyLetsCode, 2020-12-21:
yes

lorol, 2020-12-21:
Use the previous release of the library and see the platformio example.

solroshan, 2020-12-23:
Hello geezers,
I got an error looks like yours, here it is;

[image]
Here my .ini details;
[image]

Checked the "esp_littlefs.c" for uncomment  the line
#define CONFIG_LITTLEFS_FOR_IDF_3_2
But this line already defined.
Also i changed the library with previous release(1.0.4)
Error is constant for now :(

I'll come back here If I find a way to fix. Ciao..

solroshan, 2020-12-23:
> Hi
> Uncomment the line
> #define CONFIG_LITTLEFS_FOR_IDF_3_2
> in esp_littlefs.c.

Yuppie! Its worked. I checked 2nd time the #define idf3.2 line and error gone away permanently.
Thank you @lorol

ndastur, 2021-01-22:
I think it better to define this in ```platformio.ini```

```
build_flags =
    -DCONFIG_LITTLEFS_FOR_IDF_3_2
```