lorol/LITTLEFS #16: Platformio and littlefs

jackjansen, 2021-01-03:
At the moment it is rather nasty to install the correct version of Arduino and littlefs when using platformio.

The core issue is #14: littlefs by default is compatible with a newer version of IDF than platformio installs for Arduino, therefore littlefs 1.0.5 does not compile.

The work-around of manually editing the littlefs source does not work in case of using platformio to build with a CI/CD (there's no-one there to do the manual editing).

At the moment my workaround is to use 

```
lib_deps = lorol/LittleFS_esp32@!=1.0.5
```

but in effect this uses 1.0, because intermediate versions (like 1.0.4) are not available in the platformio library manager.

Life would be a lot simpler if littlefs set this `CONFIG_LITTLEFS_FOR_IDF_3_2` define dynamically (at compile time), by triggering on some IDF define for the IDF version currently being built for. <https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/system.html#idf-version-h> suggests this is possible (but I've never tried it myself).

lorol, 2021-01-04:
Hi For Arduino I already did by best to recognize IDF at compile time and the last repo code works on any.
I don't know where exactly PIO is in different IDF's support.
There is nothing significant changed (other than timestamps) since the very initial version and you can use it in PIO.

Note, I will not maintain too much this detached library as it is already part of next Arduino core major release (2.xx) based on IDF 4.2. Then everything is up to PIO people. 

Also, you can locally copy the last source to your PIO project /lib folder and use it as you wish. It should work.

jackjansen, 2021-01-04:
I _think_ that the code you referenced does exactly what I need. Will investigate and report back.

Jason2866, 2021-01-15:
Why not using the git commit you want with PlatformIO?
Something like `lib_deps = https://github.com/lorol/LITTLEFS.git#11d47dffd086c2de4ea9d7d375f29244a6b83488`

jackjansen, 2021-01-15:
> Why not using the git commit you want with PlatformIO?

I don't like that, too brittle (especially with CI/CD based building). Given the comments by @lorol I've stopped using LittleFS on esp32 for now, reverting to SPIFFS. As soon as PlatformIO use IDF 4.2 and the corresponding Arduino as that standard install it'll all work.

lorol, 2021-01-15:
Does it take the latest code this way? (w/ #git commit in URL).
Also:
>....
> but in effect this uses 1.0, because intermediate versions (like 1.0.4) are not available in the platformio library manager.
> ...

@jackjansen  and  @Jason2866 
Can someone confirm this, please?  It looks strange for me that only major versions are considered, but if it makes your life easier I may just "release" a major version 2 (I wanted to associate somehow 1.x to the esp32 Arduno core release versioning but anyway ...)

I really don't know how platformio library manager works, neither I recall submitting something myself there. 
:)  first time I see this, did you try the recommendations?:
https://platformio.org/lib/show/11164/LittleFS_esp32/installation

Jason2866, 2021-01-15:
Release 1.0.4 is commit https://github.com/lorol/LITTLEFS/commit/9b67677007a8f930e21ef1ff1da5aafc9a0f93dc
so the platformio entry is
`lib_deps = https://github.com/lorol/LITTLEFS/commit/9b67677007a8f930e21ef1ff1da5aafc9a0f93dc`

Jason2866, 2021-01-15:
Btw upcoming Arduino ESP 1.0.5 release is based on IDF3.3.
Dont expect a Arduino ESP32 release based on IDF4.X in the near future

lorol, 2021-01-15:
> 
> 
> Btw upcoming Arduino ESP 1.0.5 release is based on IDF3.3.
> Dont expect a Arduino ESP32 release based on IDF4.X in the near future

It will be earlier than you think. ESP want to have esp32s2 Arduino support, so If you look at repository, both are updating together:
master and 4.2. (No any core release for IDF in between, all other branches are abandon) I hope PIO will realize this :) sooner.

Jason2866, 2021-01-15:
We will see 😀  Using new releases is now easy with Platformio. No more waiting for Platformio release. With this [commit](https://github.com/espressif/arduino-esp32/commit/d2530850a38af18ada5e805f0f4539e0048c43b3) the Espressif release can be used directly from Platformio

lorol, 2021-02-09:
Closing ... 
Feel free to add updates on this topic.

closed 2021-02-09 by lorol