lorol/LITTLEFS #17: LFS_NAME spiffs override for Arduino library

tsctrl, 2021-01-11:
Hi @lorol,

`static constexpr const char LFS_NAME[] = "spiffs";`

wondering why the partition name was defaulted to spiffs and how to override or add more values in the sketch?
i think this could be any value that user set in the partition table.

Thank you

lorol, 2021-01-11:
The main reason was to re-use uploading tool and remain like this.
See here: https://github.com/lorol/arduino-esp32fs-plugin/blob/master/src/ESP32FS.java#L274
If you you are using your own string, then the uploading add-on for Arduino IDE will not work.

tsctrl, 2021-01-12:
Hi @lorol,

understood, the partition label/name itself can be define with custom value. this is to have better control when selecting/loading different partition to read.

# ESP-IDF Partition Table
# Name,   Type, SubType, Offset,  Size, Flags
anyvalue1,  data, spiffs,     0x4000,  1M,
anyvalue2,  data, spiffs,     0x8000,  1M,

refer to the existing SPIFFS library implementation here: [https://github.com/espressif/arduino-esp32/blob/2452c1fb539246e47a715b74a3ad25b8a7ebbec7/libraries/SPIFFS/src/SPIFFS.h#L27](SPIFFS.h#L27)

the SPIFFS library allow us to select different partition label to load by specifying partitionLabel parameter. if this existing SPIFFS library works on the upload tools than probably the partition label was actually set somewhere during the library load. however for littleFS the label was set as constant in the library itself so that limit the possibility for us to use different partition label. I believe that the upload library should be able to read all available partition to write instead of using predefined values.

to note here, the partition label is any value that can be set and not always 'spiffs'. but not the type and subtype. i think you are referring partition scheme to a type/subtype but not the partition label. however, the constant that was in the LITTLEFS, i believe was the partition label which suppose to be a subType. correct me if i am wrong. but i did some changes to the constant to select different partition to be able to load the partition and read the content as below.

`	
#define FILESYSTEM LITTLEFS
#include <LITTLEFS.h>
`
`while (!LITTLEFS.begin(false, "/folder", 10, partition_label)) {
	if (!mount_spiffs(false, partition_label)) {
		Serial.println("SPIFFS Mount Failed");
	}
}`

`FILESYSTEM.open('/');`

i did not have good knowledge on how the upload library work and havent try those to say. let me know if there is approach to load a partition before reading its content other then the one i use above.

lorol, 2021-01-12:
Hi @tsctrl 

Please submit a PR with changes with what you believe is necessary. It is feasible the label to be used as a parameter.
However keep in mind:
- If you propose a name for the default value, different than "spiffs" then you should submit a PR to the core to include dedicated partition schemes for LittleFS with your default label name.
- Before next major esp32 core release, 2.x it will not be considered, I believe

The main purpose of all this library was to bring to Arduino the robustness of new filesystem as simple, straightforward possible for general usage.

Of course I will keep your suggestions in mind.
Please test, provide some sketches with usage of 2 x data partitions of same type etc.

Thank you,

closed 2021-02-17 by tsctrl