lorol/LITTLEFS #18: LittleFS unresponsive under many simultaneous .open() / read() for the same file

raomin, 2021-01-31:
Hi,

I have a project using AsyncWebServer in a captive portal mode and sending all requests to `/index.html`.
When I connect my PC to the ESP32 wifi, it generates many random requests that all ends up in requests for `/index.html` (captive portal).
On SPIFFS I had no issues to get these requests served (the file itself is very small).
After moving to LittleFS, AsyncWebServer sends the header but cannot send the file content and the page hangs.
If I replace the automatic redirect to `/index.html` by a `404`, `/index.html` get served without any issue when I request it.

I did not dig in LittleFS_esp32 code too much, but I see it is using semaphores. Could there be a race condition when multiple `read()` requests come _at the same time_?

lorol, 2021-02-01:
Please share the code of your example that issue can be reproduced.
Also check at https://github.com/joltwallet/esp_littlefs if you can find anything related.
The implementation is taken from there.

lorol, 2021-04-06:
@raomin You can reopen if not resolved. Please add more details if this is the case.

closed 2021-04-06 by lorol