lorol/LITTLEFS #19: LITTLEFS.h filename case not compatible with ESP8266 version when using Linux

darrylb123, 2021-02-10:
The file names for littlefs on Esp8266 are mixed case, eg LittleFS.h etc
This version for ESP32 are all in upper case (LITTLEFS.h), making the code to use the library impossible to use on case sensitive OS such as Linux without lots of #ifdefs

lorol, 2021-02-10:
The file systems between ESP8266 and ESP32 are differently implemented anyway. 
Headers are different, and code cannot be reused directly. 
Please provide an example of your challenge.

darrylb123, 2021-02-11:
There will be many more than one #ifdef to be able to compile on both. 

Did the case change because you didn't want the APIs to be the same??
ESP8266 Example code  The FORMAT_LITTLEFS_IF_FAILED is implied.
if (!LittleFS.begin()) {
    Serial.println("LittleFS mount failed");
    return;
  }

ESP32 Example code doing the same thing. 
if(!LITTLEFS.begin(FORMAT_LITTLEFS_IF_FAILED)){
        Serial.println("LITTLEFS Mount Failed");
        return;
    }
Neither will compile on the other architecture on Linux.

darrylb123, 2021-02-11:
I've #defined the filesystems. A bit ugly but it works.
#if defined(ESP8266)
#define MYFS LittleFS
#define FORMAT_LITTLEFS_IF_FAILED true

#elif defined(ESP32)
#define MYFS LITTLEFS
#define FORMAT_LITTLEFS_IF_FAILED true
#endif
  if (!MYFS.begin(FORMAT_LITTLEFS_IF_FAILED)) {
    Serial.println("LittleFS mount failed");
    return;
  }

lorol, 2021-02-11:
Hi,
Yes, I made it different. 
I actually asked other people at one point, but no one seem to be interested in thinking.
The Arduino wrapper is very small additional work comparing to the ESP32 IDF implementation.
It may be reworked when core 2.x is coming as internal part of it as other decisions are pending, like constants controlled by Espressif, overlapping or detaching from SPIFFS completely, etc.

lorol, 2021-04-03:
I will discuss with @me-no-dev what will be the best solution for case-sensitivity of the library name and filenames on next major esp32 core release 2.x based on IDF 4.x. There the LittleFS will be integrated.
For now - closing this issue.

closed 2021-04-03 by lorol