lorol/LITTLEFS #20: How do you implement thread saftey?

MSmithDev, 2021-02-23:
I saw that Thread Safety was added back in December, Is there any documentation on this? how do you implement it?

lorol, 2021-02-23:
Not implemented. I see [what do you mean](https://github.com/littlefs-project/littlefs/blob/1863dc7883d82bd6ca79faa164b65341064d1c16/lfs.h#L183) ...

This is a wrapper of the wrapper. Perhaps @BrianPugh can take a look in future.

MSmithDev, 2021-02-23:
Dang, I guess I misinterpreted that. Thanks anyway!

closed 2021-02-23 by MSmithDev

BrianPugh, 2021-02-23:
@MSmithDev if you're curious, basically upon filesystem initialization, I just allocate a FreeRTOS `xSemaphoreCreateRecursiveMutex()` and acquire/release it around most commands. For example:

https://github.com/joltwallet/esp_littlefs/blob/d268e18d774431fb6f0e444feef4e97a2dfaaa29/src/esp_littlefs.c#L1153

I didn't know that littlefs added thread safety internally; we could migrate to that, but I don't really see a benefit at the moment.

MSmithDev, 2021-02-24:
@BrianPugh Thanks, I will definitely look into that, I personally haven't done much with Mutex and Semaphores so I need to read up on that first.

johnnytolengo, 2021-07-27:
Did anyone solve the Thread Safe issue?
I still have crashes and sometimes the FS is corrupted after that, of course using boths cores with parallel tasks.

MSmithDev, 2021-07-27:
There is no thread safety built in currently. You have to handle the tasks in your code and make sure it can't happen.

BrianPugh, 2021-07-27:
All actions are guarded by mutexs, I implemented this before upstream littlefs had support. What exact use case is causing errors?

johnnytolengo, 2021-07-27:
@BrianPugh I don't know exactly where is the colision but I seems to be when some task is writing or reading in Core0 and ESPAsyncWebServer request on Core1 accessing to some datafile.

johnnytolengo, 2021-07-27:
I am currently writting a wrapper for LITTLEFS using SemaphoreHandle_t at open() and close() functions

BrianPugh, 2021-07-27:
I think you might be misguided, this is already builtin:

https://github.com/lorol/LITTLEFS/blob/master/src/esp_littlefs.c#L662

Is your issue related to this? https://github.com/joltwallet/esp_littlefs/issues/18  Basically its an upstream issue that having two filedescriptors open for the same file will run into issues.

johnnytolengo, 2021-07-28:
OMG! I didn't know that "sem_take()" but after looking the code (esp_littlefs.c) and enabling the log with LOG_LOCAL_LEVEL=5
I see that the sem_take() is not executed using the example code: _https://raw.githubusercontent.com/lorol/LITTLEFS/master/examples/LITTLEFS_PlatformIO/src/main.cpp_

It seems that esp_littlefs.c is not used everything is linked to FS.cpp so now I am fully misunderstanding how LITTLEFS is working. 
Please if any one can tell me how I can trace and enable debugging.

Regarding the **two filedescriptors open** is possible that that is my problem, to make the code thread safe will be enough to put the sem_take() before the File pointer is linked to LITTLEFS library, is it?