lorol/LITTLEFS #22: unable to remove a file when it's open?

proddy, 2021-03-04:
With SPIFFS this was valid:
```C
  File root = fs->open("/configdir");
  File file;
  while (file = root.openNextFile()) {
    fs->remove(file.name());
  }
```

but with LITTLEFS it will throw an error `esp_littlefs: Failed to unlink path "/configdir/file.json". Has open FD.`

it's easy to circumvent by cloning the path and issuing a `file.close()` before the `remove()` but was I just curious whether this is by design?

lorol, 2021-03-04:
Yes, it is different. @BrianPugh may elaborate more details on the lower level.

lorol, 2021-03-05:
@proddy, see this issue on same subject - it was already discussed.
https://github.com/lorol/LITTLEFS/issues/7
Hope that helps you to find an acceptable solution

proddy, 2021-03-05:
ah thanks. I should have browsed the closed issues. It's not a problem, but people using Arduino and thinking they can swap out SPIFFS with LITTLEFS will likely encounter the same issue. Perhaps worth documenting it somewhere. Anyway thanks for following up and great work on the port!

closed 2021-03-05 by proddy