lorol/LITTLEFS #25: ESP32 - littlefs path prefix is added - works as designed ?

saikek, 2021-03-22:
I'm using `ESP Async WebServer@1.2.3` with `lorol/LittleFS_esp32 @ ^1.0.5`. 
I moved my project from SPIFFS to LITTLEFS.
When I'm running ESP8266 code, paths are resolved correctly.
When I'm using    ESP32 with LittleFS_esp32  library, it first tries to resolve paths starting with littlefs, for example:
```
get /folder/example.html
logs say:

/littlefs/folder/example/html does not exist
/littlefs/folder/example/html.gz does not exist
/littlefs/folder/example/html/index.html does not exist
/littlefs/folder/example/html/index.html.gz does not exist
...
OK fetch from /folder/example.html
```

Is there a configuration flag to specify prefix ? 
Is this behavior as designed by Littlefs or i'm missing something ?

saikek, 2021-03-22:
Is that a case ? 
> LittleFS has folders, you need to iterate files in folders unless you set #define CONFIG_LITTLEFS_SPIFFS_COMPAT 1
> At root a "/folder" = "folder"

lorol, 2021-04-06:
Yes, I believe this is the reason. If it doesn't help please post more details. 
Closing the issue for now.

closed 2021-04-06 by lorol