lorol/LITTLEFS #27: LITTLEFSFS::begin() unexpected behavior

robcazzaro, 2021-04-07:
Relatively minor issue. The code behaves as expected, but doesn't return the right value

I used the following code in the setup() part of my project

```
  // Format SPIFFS if not yet
  if (!FileFS.begin(true))
  {
    Serial.println(F("Init failed! AutoFormatting."));
  }
```

I expected to see the "Init failed! AutoFormatting." message on my terminal window, instead I get 

```
lib\LittleFS_esp32\lfs.c:998:error: Corrupted dir pair at {0x0, 0x1}
E (230) esp_littlefs: mount failed,  (-84)
E (234) esp_littlefs: Failed to initialize LittleFS
```

It actually auto-formats the SPIFFS partition, just doesn't return false, so the code doesn't know it's been auto formatted. In some cases, the code needs to know if the SPIFFS area has been auto formatted, so would be nice to fix

lorol, 2021-04-08:
@robcazzaro 
Please try on your example to use !FileFS.begin(false) and post what is returning.

lorol, 2021-04-09:
Hi, the messages like "Corrupted dir pair at {0x0, 0x1} E (4349)" are from lower level of the driver.
The whole thing is still in development. Maybe later these will be cleaned by adjusting the error levels there.
The lib is really the highest wrapper, similar to SPIFFS. I didn't test much the format scenarios.
For now I will leave it this way, as we are moving it as a part of core V2. The driver will be maintained at IDF side and use pre compiled as other FS

robcazzaro, 2021-04-09:
Agreed. This was mostly a misunderstanding on my side, not a real issue in your code. Thanks again for adding the multiple SPIFFS functionality

closed 2021-04-09 by robcazzaro

lorol, 2021-04-09:
@robcazzaro 
See: https://github.com/lorol/LITTLEFS/blob/master/src/lfs.c#L10 if you want to suppress or verbose more the debug messages