lorol/LITTLEFS #28: ESP32 restarts if route exist but file dont with async webserver.

zekageri, 2021-04-13:
### If i request a webpage with defined route like this:

```
server.on("/devicesTable", HTTP_GET, [](AsyncWebServerRequest *request) {
    AsyncWebServerResponse* response = request->beginResponse(LITTLEFS, "/hwConfig.html", "text/html");
    response->addHeader("Content-Encoding", "gzip");
    request->send(response);
});
```

### but the files does not exists in the filesystem, the esp restarts with the following serial message:

```
E (3608859) esp_littlefs: Failed to opendir "/hwConfig.html". Error  (-2)
E (3608873) esp_littlefs: Failed to opendir "/hwConfig.html.gz". Error  (-2)
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.        
Core 1 register dump:
PC      : 0x400d52df  PS      : 0x00060930  A0      : 0x800ee1bf  A1      : 0x3ffd7150  
A2      : 0x00000000  A3      : 0x3ffc7b78  A4      : 0x3ffe3220  A5      : 0x195f9cbd    
A6      : 0x0000000d  A7      : 0x00000000  A8      : 0x800d52d9  A9      : 0x3ffd7130  
A10     : 0x3ffd716c  A11     : 0x3f4006f2  A12     : 0x195f9cbd  A13     : 0x3ffc7b78    
A14     : 0x00000000  A15     : 0x3ffd716c  SAR     : 0x00000004  EXCCAUSE: 0x0000001c    
EXCVADDR: 0x00000000  LBEG    : 0x4008cc35  LEND    : 0x4008cc45  LCOUNT  : 0xfffffffb    

ELF file SHA256: 0000000000000000

Backtrace: 0x400d52df:0x3ffd7150 0x400ee1bc:0x3ffd71a0 0x400ebb29:0x3ffd71f0 0x400ebbfa:0x3ffd7230 0x400ebe71:0x3ffd7280 0x400e59e2:0x3ffd72a0 0x400e5a79:0x3ffd72e0 0x400e61ea:0x3ffd7300 0x4008e89a:0x3ffd7330
  #0  0x400d52df:0x3ffd7150 in std::_Function_handler<void (AsyncWebServerRequest*), HTTP_Requests()::{lambda(AsyncWebServerRequest*)#4}>::_M_invoke(std::_Any_data const&, AsyncWebServerRequest*&&) at src/Own_Headers/ModBus.h:365 (discriminator 2)
  #1  0x400ee1bc:0x3ffd71a0 in AsyncWebServer::_rewriteRequest(AsyncWebServerRequest*) at 
lib\ESPAsyncWebServer-dev\src/StringArray.h:73
  #2  0x400ebb29:0x3ffd71f0 in std::function<void (AsyncWebHeader* const&)>::function(std::function<void (AsyncWebHeader* const&)> const&) at lib\ESPAsyncWebServer-dev\src/WebRequest.cpp:806
      (inlined by) LinkedList<AsyncWebHeader*, LinkedListNode>::LinkedList(std::function<void (AsyncWebHeader* const&)>) at lib\ESPAsyncWebServer-dev\src/StringArray.h:64
      (inlined by) AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*) at lib\ESPAsyncWebServer-dev\src/WebRequest.cpp:70
  #3  0x400ebbfa:0x3ffd7230 in AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*) at lib\ESPAsyncWebServer-dev\src/WebRequest.cpp:806
  #4  0x400ebe71:0x3ffd7280 in LinkedList<AsyncWebParameter*, LinkedListNode>::free() at lib\ESPAsyncWebServer-dev\src/WebRequest.cpp:806
      (inlined by) AsyncWebServerRequest::~AsyncWebServerRequest() at lib\ESPAsyncWebServer-dev\src/WebRequest.cpp:83
  #5  0x400e59e2:0x3ffd72a0 in std::function<void (void*, AsyncClient*)>::~function() at 
lib\AsyncTCP-Fix_-111\src/AsyncTCP.cpp:1080
      (inlined by) AsyncClient::~AsyncClient() at lib\AsyncTCP-Fix_-111\src/AsyncTCP.cpp:596
  #6  0x400e5a79:0x3ffd72e0 in AsyncClient::_s_fin(void*, tcp_pcb*, signed char) at lib\AsyncTCP-Fix_-111\src/AsyncTCP.cpp:1080
  #7  0x400e61ea:0x3ffd7300 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count(std::__shared_count<(__gnu_cxx::_Lock_policy)2> const&) at C:\Users\Pc\.platformio\packages\framework-arduinoespressif32\libraries\FS\src/FS.cpp:258
      (inlined by) std::__shared_ptr<fs::FileImpl, (__gnu_cxx::_Lock_policy)2>::__shared_ptr(std::__shared_ptr<fs::FileImpl, (__gnu_cxx::_Lock_policy)2> const&) at c:\users\pc\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits/shared_ptr_base.h:923
      (inlined by) std::shared_ptr<fs::FileImpl>::shared_ptr(std::shared_ptr<fs::FileImpl> const&) at c:\users\pc\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits/shared_ptr.h:107
      (inlined by) fs::File::File(std::shared_ptr<fs::FileImpl>) at C:\Users\Pc\.platformio\packages\framework-arduinoespressif32\libraries\FS\src/FS.h:50
  #8  0x4008e89a:0x3ffd7330 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1) 

Rebooting...
ets Jul 29 2019 12:21:46
```

**I have an onNotFound handler too and the "NotFound.html" does exists on the filesystem.**

```
server.onNotFound([](AsyncWebServerRequest *request){
    AsyncWebServerResponse* response = request->beginResponse(LITTLEFS, "/NotFound.html", "text/html");
    response->addHeader("Content-Encoding", "gzip");
    request->send(response);
 });
```

lorol, 2021-04-15:
Hi @zekageri 
Thank you for pointing it out. 
Could you please check what is the behavior with SPIFFS and FatFS?
This is an AsyncWebServer related issue (what version/clone?) and depends of too many things. Async lib, core, FS, 
I may not be able to help you much better than your workaround.
File / Dir are differently handled vs. SPIFFS and as you know when AsyncWebServer was developed by me-no-dev LittleFS was not yet implemented at all. All "changes in clones" of AsyncWebServer afterwards were done per individual needs.

zekageri, 2021-04-15:
Thanks for the response. I will check it with SPIFFS.

lorol, 2021-04-15:
Try also with my clone of AsyncWebServer.