lorol/LITTLEFS #32: Creating lfs_config to use lfs_file_write() to write to file (help wanted)

johnsmakar, 2021-05-29:
Hello, After doing a bit of reading over the source code, I'd like to use the lfs_file_write() function from `lfs.c` to write to file because it handles caching a buffer for me to reduce writes to flash. I am trying to create a lfs_config struct but cannot figure out some fields.
```
const struct lfs_config cfg = {
    // block device operations
    .read  = user_provided_block_device_read,
    .prog  = user_provided_block_device_prog,
    .erase = user_provided_block_device_erase,
    .sync  = user_provided_block_device_sync,
```
I am not sure how to ago about filling `.read` `.prog` `.erase` `.sync` . I see that in in `esp_littlefs.c` 
```
        // block device operations
        efs->cfg.read  = littlefs_api_read;
        efs->cfg.prog  = littlefs_api_prog;
        efs->cfg.erase = littlefs_api_erase;
        efs->cfg.sync  = littlefs_api_sync;
```
littlefs_api is used but I get a ton of errors like:
`src\main.cpp:61:1: sorry, unimplemented: non-trivial designated initializers not supported`
when trying to implement something similar. Any help would be much appreciated!

lorol, 2021-05-29:
Hi @johnsmakar 
Again @BrianPugh may be the best source of help. 
However, please explain what exactly you want to reach by going lower level.
It may be feasible differently.
Regards

BrianPugh, 2021-05-29:
hey @johnsmakar

to echo @lorol 's response, what goal exactly are you trying to reach? The point of the whole driver is to abstract away the lower level filesystem actions. You can configure the automatic buffer size in the config system. Writes won't be performed until either the buffer is full, or a `sync` call is made. You can add some logging to the explicit chip writes if you want to verify its writing-to-chip when you intend.

johnsmakar, 2021-05-29:
Thanks for the response @BrianPugh

The system I am designing is writing at 24fps and I don't and to write to flash at 24 times a second. The example which includes:
```
void writeFile(fs::FS &fs, const char * path, const char * message){
    Serial.printf("Writing file: %s\r\n", path);

    File file = fs.open(path, FILE_WRITE);
    if(!file){
        Serial.println("- failed to open file for writing");
        return;
    }
    if(file.print(message)){
        Serial.println("- file written");
    } else {
        Serial.println("- write failed");
    }
    file.close();
}
```
uses `file.print` which looks to write to flash immediately, and lfs_file_write() does like what you said, "Writes won't be performed until either the buffer is full, or a sync".

Can you clarify which write function are you referring to and where the config system is?

Thanks

Can you clarify what config system

BrianPugh, 2021-05-29:
[this function](https://github.com/lorol/LITTLEFS/blob/f3bce17b028378d994d84d7a2c105dc11fcfdf27/src/littlefs_api.c#L30) is called whenever an actual write-to-chip is performed

johnsmakar, 2021-05-29:
but what function can I use to "write" and will handle writing to buffer until it is full or .sync is called? littlefs_api_prog just writes to chip.

BrianPugh, 2021-05-30:
if you just use the POXIS interface from the esp virtual filesystem, it's all handled for you.

johnsmakar, 2021-05-30:
I apologize of for my unfamiliarity with POXIS interface from the esp virtual filesystem, I tried to do a little research, but you are suggesting using [VFS](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/vfs.html) instead of LFS?

And if so is there no easy way to stay in lfs and use lfs_file_write?

BrianPugh, 2021-05-30:
VFS is just an abstraction from the lower level filesystem (which this repo, LFS, implements). The whole point of this repo is to "install" the filesystem like the examples, then interact with it via the standard POSIX interface which will handle all of the buffering and other lower level tasks for you. Basically, if you follow the examples, I'm sure it will automatically accomplish what your trying to achieve.

johnsmakar, 2021-06-08:
I am sorry, but I am still a little confused. I followed the example to this repo, and it does not seem to accomplish what I am trying to achieve.

I am trying to reduce the number of writes to flash by using a buffer, I see that the lower level functions doing that but when I create traces in my program and call `appendFile` a function defined in the example and read the file immediately after (before calling close or sync) I see the data was written even though it did not fill the block size buffer. I hope this makes sense.

I apologize for any misunderstandings I have, but if you could provide me with some example code that reflects what I am try to achieve it would be much appreciated

johnsmakar, 2021-06-08:
I am sorry, but I am still quite confused. I followed the example to this repo, and it does not seem to accomplish what I am trying to achieve.

I am trying to reduce the number of writes to flash by using a buffer, I see that the lower level functions doing that but when I create traces in my program and call `appendFile` a function defined in the example and read the file immediately after (before calling close or sync) I see the data was written even though it did not fill the block size buffer. I hope this makes sense.

I apologize for any misunderstandings I have, but if you could provide me with some example code that reflects what I am try to achieve it would be much appreciated

BrianPugh, 2021-06-08:
Whenever a read is performed, the write buffer is first flushed-to-chip.

johnsmakar, 2021-06-08:
I see, thanks.

Is there any way to monitor the buffer? for example using `ESP.getFreePsram()` to check if it is caching in memory? or does it not work that way?

BrianPugh, 2021-06-08:
I don't believe there is. [Can you outline your actual use case](https://xyproblem.info)? I feel like perhaps there might be a better way to go about it. If there isn't a better alternative, I can give you some guidance to lower-level use.

johnsmakar, 2021-06-08:
Haha, Thanks for that resource link, that's exactly what is happening!

I have ESP which is connected to an IMU and Encoders, that data which is being read from the sensor and encoder is being streamed over TCP to another machine at 24fps. I am using the second core of the ESP as a 'logger' task to log the data in a .csv file then when the session is over to send that file over TCP to the device as well. (I am currently using `QueueHandle_t` from FreeRTOS to send data between the 2 cores)

What is the best approach to frequently write to a file without killing the lifespan of the flash?

BrianPugh, 2021-06-08:
are  you using Arduino, or just esp-idf?

In either case, just have the logger task write the data as it comes in, and then do a read at the end. Don't attempt to read back after each write or anything. This _should_ result in the least number of actual writes-to-chip.

johnsmakar, 2021-06-08:
I am using the Arduino framework on a TinyPICO board... Sounds good.

I am just curious where is the buffer created? in SPRAM or FLASH?

Thanks for all the help!

BrianPugh, 2021-06-08:
do you mean like external ram (PSRAM if your hardware has it) or internal ram? I **think** its allocated just by a vanilla malloc, so I don't think theres any strong guarentees where it goes. However, its been a while since i worked on esp stuff.

johnsmakar, 2021-06-08:
Yes the SPRAM is external which I believe requires `ps_malloc` for Arduino. I believe the internal SRAM is faster anyways.

Again I appreciate all the time and help!

closed 2021-06-08 by johnsmakar