lorol/LITTLEFS #33: Append to file after removing last character from previous write or append

happytm, 2021-06-11:
I am using your excellent library and everything works as expected except when I try to append I wanted to remove last character from previous append and it is not happening.

My code is below:
```
#include <Arduino.h>
#include "LITTLEFS.h"

void setup() {
  Serial.begin(112500);
  delay(1000);

  LITTLEFS.begin(true);
  File f = LITTLEFS.open("/test.json", "w");
  f.println("[1623287518,24785,1535,15,6,-22,292,49,89,250,76]");
  f.close();
  Serial.println("Wrote first line to file");

  f = LITTLEFS.open("/test.json", "a");
  Serial.print("File size: "); Serial.println(f.size());
  f.seek((f.size() - 1), SeekSet);
  Serial.print("Position: "); Serial.println(f.position());
  f.println(",1623287518,24785,1535,15,6,-22,292,49,89,250,76]");
  Serial.println();
  Serial.println("Appended to file");
  Serial.print("File size: "); Serial.println(f.size());
  f.close();
   
  f = LITTLEFS.open("/test.json", "r");
  Serial.println("Reading from file...");
  while(f.available()){
  Serial.write(f.read());
  }
 f.close();
}

void loop() {}
```
And result I get is below:
```
Wrote first line to file
File size: 51
Position: 50

Appended to file
File size: 51
Reading from file...
[1623287518,24785,1535,15,6,-22,292,49,89,250,76]
,1623287518,24785,1535,15,6,-22,292,49,89,250,76]
```
But result I wanted to see was below:
```
[1623287518,24785,1535,15,6,-22,292,49,89,250,76
,1623287518,24785,1535,15,6,-22,292,49,89,250,76]
```
May be I am doing it all wrong.

Can you show me the right way to do it?

Thanks.

BrianPugh, 2021-06-11:
`println` adds an invisible carriage return to the end of your string, so try just using `print`. Also, SeekEnd is a bit more convenient in your use case.

happytm, 2021-06-11:
Brian,

Thank you for your prompt reply.

I tried both things you suggested but still I can not get rid of square bracket at end of first line.

lorol, 2021-06-11:
@happytm 
I would not mix appending and backing one position at same time, as it looks a bit tricky in general:
https://stackoverflow.com/questions/29013495/opening-file-in-append-mode-and-seeking-to-start
Classically, you ether rewrite the whole file or append at the end.
I would buffer and rewrite. If you are concerned you don't have enough RAM to do so, then you will be in a trouble anyway.

happytm, 2021-06-11:
@lorol Thank you for the link.

happytm, 2021-06-12:
> Classically, you ether rewrite the whole file or append at the end.<

Writing whole file when I want to add one row every few minutes is looking little inefficient.

Is there any way around you can think of  to append at end of file - 1.

Thanks.

lorol, 2021-06-12:
Probably not. 
Try to open in r+ mode instead of a.

happytm, 2021-06-12:
@lorol Thanks man for your help. It works as expected with r+. 

```
#include <Arduino.h>
#include "LITTLEFS.h"

void setup() {
  Serial.begin(112500);
  delay(1000);

  LITTLEFS.begin(true);
  File f = LITTLEFS.open("/test.json", "w");
  f.print("[1623287518,24785,1535,15,6,-22,292,49,89,250,76]");
  f.close();
  Serial.println("Wrote first line to file");

  f = LITTLEFS.open("/test.json", "r+"); // See https://github.com/lorol/LITTLEFS/issues/33
  Serial.print("File size: "); Serial.println(f.size());
  f.seek((f.size()-1), SeekSet);
  Serial.print("Position: "); Serial.println(f.position());
  f.print(",1623287620,24786,1635,16,26,-34,285,56,78,245,82]");
  Serial.println();
  Serial.println("Appended to file");
  Serial.print("File size: "); Serial.println(f.size());
  f.close();
   
  f = LITTLEFS.open("/test.json", "r");
  Serial.println("Reading from file...");
  while(f.available()){
  Serial.write(f.read());
  }
 f.close();
}

void loop() {}
```