lorol/LITTLEFS #43: LITTLEFSimpl::open bug with latest Platformio (ES32 arduino framework) in Visual Studio Code

har-in-air, 2021-10-15:
```
Compiling .pio/build/esp32dev/lib63a/LittleFS_esp32/LITTLEFS.cpp.o
.pio/libdeps/esp32dev/LittleFS_esp32/src/LITTLEFS.cpp: In member function 'virtual bool LITTLEFSImpl::exists(const char*)':
.pio/libdeps/esp32dev/LittleFS_esp32/src/LITTLEFS.cpp:44:28: error: no matching function for call to 'LITTLEFSImpl::open(const char*&, const char [2])'
     File f = open(path, "r");
In file included from .pio/libdeps/esp32dev/LittleFS_esp32/src/LITTLEFS.cpp:17:
/home/hari/.platformio/packages/framework-arduinoespressif32/libraries/FS/src/vfs_api.h:38:17: note: candidate: 'virtual fs::FileImplPtr VFSImpl::open(const char*, const char*, bool)'
     FileImplPtr open(const char* path, const char* mode, const bool create) override;
                 ^~~~
/home/hari/.platformio/packages/framework-arduinoespressif32/libraries/FS/src/vfs_api.h:38:17: note:   candidate expects 3 arguments, 2 provided
*** [.pio/build/esp32dev/lib63a/LittleFS_esp32/LITTLEFS.cpp.o] Error 1
```
Fixed by adding third argument
File f = open(path, "r", false);

goblingift, 2022-05-27:
@tonyrewin - thanks for your bugfix. When will this gets released?

Michael2MacDonald, 2022-09-28:
Screw it. I am going to submit a fix today if possible.

Michael2MacDonald, 2022-09-28:
For some reason, I can't get the PR I created to show up here. If someone could help teach me how to do that or why it is not working then that would be great.

But my PR can be found here: https://github.com/lorol/LITTLEFS/pull/56

Please review and merge my PR so this tiny issue can finally be over!

waymond91, 2022-11-12:
Im still running into this using platformio. Am I installing the wrong version?

eatonJ, 2023-04-04:
FYI ... I had this issue today and used this fix to correct the error . Thanks

sommerper, 2023-04-17:
@eatonJ How did you fix it? I'm still getting the error when importing to PlatformIO. :-/
Latest release seems to be `lorol/LittleFS_esp32@^1.0.6`.

zhaocundang, 2023-04-27:
yes, 2023 5-1 still error

dharms87, 2023-05-06:
Tried to implement this library in a PlatformIO project. Received similar error messages posted above.

However, I noticed this message in the output when compiling:
LittleFS_esp32/src/esp_littlefs.c:19:2: warning: #warning ("Use the built-in LITTLEFS library") [-Wcpp]
 #warning("Use the built-in LITTLEFS library") 

After reviewing the "Built-In" portion of the supported libraries within PlatformIO. I noticed that LittleFS is built into the framework-arduinoespressif32.

My project compiled successfully after doing the following in my main.cpp file:
#include <LittleFS.h>

Then add the following in the platformio.ini file, under the lib_deps = LittleFS @ ^2.0.0

gatti-d, 2023-06-20:
Received similar error messages posted above. 
The solution proposed by #dharms87 worked as well.
Thank you!

urs8000, 2023-10-15:
still had the error compiling LITTLEFS_test.ino under Arduino 1.8.19
corrected as in first comment with ... in LITTLEFS.cpp
`bool LITTLEFSImpl::exists(const char* path)
{
    File f = open(path, "r", **false**);
    return (f == true);
}`