lorol/LITTLEFS #59: error: no matching function for call to 'LITTLEFSImpl::open(const char*&, const char [2])'

jeiry, 2023-03-03:
board: esp32s3 n8r8
psram: opi psram
flash model: QIO 120MHz
flash size: 8MB(64MB)

`#include <Arduino.h>
#include "FS.h"
#include <LITTLEFS.h>
void setup() {
  Serial.println("Mount LittleFS");
  if (!LITTLEFS.begin()) {
    Serial.println("LittleFS mount failed");
    return;
  }
  LITTLEFS.end();
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}`

Error:
`e:\files\Arduino\libraries\LITTLEFS-1.0.6\src\LITTLEFS.cpp: In member function 'virtual bool LITTLEFSImpl::exists(const char*)':
e:\files\Arduino\libraries\LITTLEFS-1.0.6\src\LITTLEFS.cpp:44:28: error: no matching function for call to 'LITTLEFSImpl::open(const char*&, const char [2])'
     File f = open(path, "r");
                            ^
In file included from e:\files\Arduino\libraries\LITTLEFS-1.0.6\src\LITTLEFS.cpp:17:
F:\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\FS\src/vfs_api.h:38:17: note: candidate: 'virtual fs::FileImplPtr VFSImpl::open(const char*, const char*, bool)'
     FileImplPtr open(const char* path, const char* mode, const bool create) override;
                 ^~~~
F:\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\FS\src/vfs_api.h:38:17: note:   candidate expects 3 arguments, 2 provided

exit status 1

Compilation error: exit status 1`

What is wrong with me? Help Please!Thank you

Riasann, 2023-03-10:
Seams its a issue with the new Version 1.0.6, i go back to Version 1.0.5 and its works again.

haklein, 2024-04-13:
getting the same error when trying to build with PIO for M5stack Core2. Specifying 1.0.5 in `lib_deps` didn't help..

Horashi0, 2024-06-17:
For anyone who has the same problem

I fixed this by downloading the 1.0.5 from here: https://github.com/lorol/LITTLEFS/releases

I then extracted the folder and renamed it to "LittleFS_esp32" and my platformio on debian recognised it and compiled without this issue ;)

g1ra, 2025-01-31:
library is now part of [Arduino esp32 core v2](https://github.com/espressif/arduino-esp32/tree/master/libraries/LittleFS)