lorol/LITTLEFS #64: Error on Arduino IDE 2.1.1 after upgrade of ESP32 version 2.0

CNSNGopenSourceProjects, 2023-07-19:
The code that was compiling yesterday (17/july/2023), stop compile today after I accept the upgrade of ESP32 version 2.0.
Please, see below the warning and errors I got when compiling for DOIT ESP32 DEVKIT V1 on my windows 10 pro computer.
I am using:
- name=LittleFS_esp32
- version=1.0.6

`
c:\Users\Projetos\Documents\Arduino\libraries\LittleFS_esp32\src\esp_littlefs.c:19:2: warning: #warning ("Use the built-in LITTLEFS library") [-Wcpp]
 #warning("Use the built-in LITTLEFS library")
  ^~~~~~~
c:\Users\Projetos\Documents\Arduino\libraries\LittleFS_esp32\src\LITTLEFS.cpp: In member function 'virtual bool LITTLEFSImpl::exists(const char*)':
c:\Users\Projetos\Documents\Arduino\libraries\LittleFS_esp32\src\LITTLEFS.cpp:44:28: error: no matching function for call to 'LITTLEFSImpl::open(const char*&, const char [2])'
     File f = open(path, "r");
                            ^
In file included from c:\Users\Projetos\Documents\Arduino\libraries\LittleFS_esp32\src\LITTLEFS.cpp:17:
C:\Users\Projetos\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.10\libraries\FS\src/vfs_api.h:38:17: note: candidate: 'virtual fs::FileImplPtr VFSImpl::open(const char*, const char*, bool)'
     FileImplPtr open(const char* path, const char* mode, const bool create) override;
                 ^~~~
C:\Users\Projetos\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.10\libraries\FS\src/vfs_api.h:38:17: note:   candidate expects 3 arguments, 2 provided
c:\Users\Projetos\Documents\Arduino\libraries\LittleFS_esp32\src\LITTLEFS.cpp: In member function 'bool fs::LITTLEFSFS::begin(bool, const char*, uint8_t, const char*)':
c:\Users\Projetos\Documents\Arduino\libraries\LittleFS_esp32\src\LITTLEFS.cpp:81:5: warning: missing initializer for member 'esp_vfs_littlefs_conf_t::dont_mount' [-Wmissing-field-initializers]
     };
     ^
`

I appreciate for any comment.

Best regards,

Fernando

LovelyDev829, 2023-07-19:
Hello Fernando,

It looks like the upgrade to ESP32 version 2.0 has caused some issues with the LittleFS_esp32 library you are using. The warnings and errors you received indicate that there are some compatibility issues between the library and the new version of ESP32.

One thing you can try is to update the LittleFS_esp32 library to a newer version that is compatible with ESP32 version 2.0. You can check the library's documentation or website to see if there are any updates available.

If updating the library doesn't work, you may need to look for an alternative library that is compatible with ESP32 version 2.0.

I hope this helps! Let me know if you have any other questions.

CNSNGopenSourceProjects, 2023-07-19:
Thank you for the reply.
I also believe there is some sort of incompatibility with the ESP32 version 2.0 package, but the LittleFS version I am using is the last one (2021). 
May be the solution is going back to the SPIFFS, isn't it?

LovelyDev829, 2023-07-19:
Yes, if you are experiencing compatibility issues with the LittleFS library and the ESP32 version 2.0 package, one solution could be to switch back to using SPIFFS (SPI Flash File System).