lorol/LITTLEFS #69: Version 1.0.6. no matching function for call to 'LITTLEFSImpl::open(const char*&, const char [2])'

VolodymyrBaisa, 2024-07-18:
Hi All.
I got issue when I start using 1.0.6 version. No issue with 1.0.5 version.
Error message:

> C:\Users\user\OneDrive\Documents\Arduino\libraries\LittleFS_esp32\src\LITTLEFS.cpp: In member function 'virtual bool LITTLEFSImpl::exists(const char*)':
> C:\Users\user\OneDrive\Documents\Arduino\libraries\LittleFS_esp32\src\LITTLEFS.cpp:44:18: error: no matching function for call to 'LITTLEFSImpl::open(const char*&, const char [2])'
>    44 |     File f = open(path, "r");
>       |              ~~~~^~~~~~~~~~~
> In file included from C:\Users\user\OneDrive\Documents\Arduino\libraries\LittleFS_esp32\src\LITTLEFS.cpp:17:
> C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\FS\src/vfs_api.h:37:15: note: candidate: 'virtual fs::FileImplPtr VFSImpl::open(const char*, const char*, bool)'
>    37 |   FileImplPtr open(const char *path, const char *mode, const bool create) override;
>       |               ^~~~
> C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\FS\src/vfs_api.h:37:15: note:   candidate expects 3 arguments, 2 provided
> 
> exit status 1
> 
> Compilation error: exit status 1

My code:
```
#include "Stream.h"
#include "storage.h" 

Storage::Storage(){}

bool Storage::mount() {
  if (!LITTLEFS.begin(FORMAT_LITTLEFS_IF_FAILED)) {
        Serial.println("LITTLEFS Mount Failed");
        return false;
    }
    return true;
}

bool Storage::readWifiConfig(String &ssid, String &password) {
    File file = LITTLEFS.open(wifiConfigFile, FILE_READ, false);
    if (!file) {
        Serial.println("Failed to open WiFi config file");
        return false;
    }

    // Allocate a temporary JsonDocument
    StaticJsonDocument<256> doc;

    // Deserialize the JSON document
    DeserializationError error = deserializeJson(doc, file);
    if (error) {
        Serial.print(F("deserializeJson() failed: "));
        Serial.println(error.f_str());
        file.close();
        return false;
    }

    ssid = doc["ssid"].as<String>();
    password = doc["password"].as<String>();

    file.close();
    return true;
}
```
Please advice.

lesept777, 2024-08-22:
I had the same problem and corrected it by changing the library
`File f = open(path, "r", false);`
in file LITTLEFS.cpp line 44 (add argument false)

48832668, 2024-12-04:
c:/users/lovefinderme/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio\build\esp32-s3-devkitc-1\libfe3\libLittleFS_esp32.a(esp_littlefs.c.o):(.literal.esp_littlefs_init+0x1c): undefined reference to `g_rom_flashchip'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32-s3-devkitc-1\firmware.elf] Error 1

it still faild , see log :(

> I had the same problem and corrected it by changing the library `File f = open(path, "r", false);` in file LITTLEFS.cpp line 44 (add argument false)

48832668, 2024-12-04:
> c:/users/lovefinderme/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: [...]

env : PIO , mcu model:  ESP32-S3 (N8R2)

48832668, 2024-12-04:
> env : PIO , mcu model: ESP32-S3 (N8R2)

code:
`#include <Arduino.h>
#include <SPI.h>
#include <Wire.h>
#include <TFT_eSPI.h>
#include <String.h>
#include <FS.h>
#include <LITTLEFS.h>

#define LEDB 13
#define LEDG 14
#define LEDR 15

#define LCD_EN 35

/*创建TFT屏幕实例*/
TFT_eSPI tft = TFT_eSPI();

/*串口交换缓冲区*/
String str_buff_pc;
String str_buff_lora;
int str_lora_read_flag = 1;

void setup()
{
  /*串口初始化*/
  Serial.begin(115200);
  Serial.println("Hello LiHuiWen!");
  Serial1.begin(115200, SERIAL_8N1, 41, 42);

  /*三色LED状态灯初始化*/
  pinMode(LEDB, OUTPUT);
  pinMode(LEDG, OUTPUT);
  pinMode(LEDR, OUTPUT);
  digitalWrite(LEDB, LOW);
  digitalWrite(LEDG, LOW);
  digitalWrite(LEDR, LOW);

  /*文件系统初始化*/
  if (!LITTLEFS.begin())
  {
    Serial.println("LittleFS init done");
  }
  else
  {
    Serial.println("LittleFS init faild!");
  }

  /*屏幕初始化*/
  tft.init();
  tft.setRotation(1);
  tft.fillScreen(TFT_BLACK);
  tft.setTextColor(TFT_WHITE, TFT_BLACK);
  tft.setTextSize(2);
  tft.setCursor(5, 5);
  tft.println("Hello World!");

}

void loop()
{

  /*监测Serial1是否有新的串口数据到达 , 有则立即加入lora字符串缓冲区*/
  while (Serial1.available() && str_lora_read_flag)
  {
    str_buff_lora += (char)Serial1.read();
  }

}
`

mjuliand, 2025-02-08:
> I had the same problem and corrected it by changing the library `File f = open(path, "r", false);` in file LITTLEFS.cpp line 44 (add argument false)

Using Arduino IDE 2.3.4 I was gettin the compilation error, but solved the problem adding "false" as the third parameter in function open.